modelica/ModelicaStandardLibrary #2477: False color of blocks in Modelica.ComplexBlocks

tobolar, 2018-02-28:
Regarding the color of the interfaces defined in `Modelica.ComplexBlocks.Interfaces`, e.g. ComplexInput, the common color of ComplexBlocks is light blue, in RGB: {85,170,255}. The icon style guide Modelica.UsersGuide.Conventions.Icons states that "The main icon color of a component shall be the same for all components of one library." But all blocks of ComplexBlocks use a base class Modelica.Blocks.Icons.Block, i.e. that one with dark blue border.

There should be defined a new particular base class for ComplexBlocks and used across the library.

tobolar, 2018-03-06:
Due to similarity to #2486, the milestone should be the same as well.

beutlich, 2018-08-06:
> Due to similarity to #2486, the milestone should be the same as well.

#2486 is fixed for MSL 3.2.3.

beutlich, 2018-10-26:
@tobolar Can you prepare a PR on this one? Thanks.

tobolar, 2018-10-29:
Will go for it.

PR #2823 (2019-02-19): Fix icon frame color for Complex blocks
close #2477
closed 2019-02-19 by beutlich