modelica/ModelicaStandardLibrary #2478: Icon style guide: missing information on libraries' colors

tobolar, 2018-02-28:
The icon style guide Modelica.UsersGuide.Conventions.Icons states that "The main icon color of a component shall be the same for all components of one library."
Even if there are Libraries' officers knowing which color has be used for new components, I would nontheless suggest to also collect an information of what color(s) has to be used for a particular library.

Based on colors of connectors collected in each `Interfaces` subpackage, I have collected the following colors:
- Modelica.Blocks: {0,0,127}
- Modelica.ComplexBlocks: {85,170,255}
- Modelica.StateGraph: {0,0,0}
- Modelica.Electrical.Analog: {0,0,255}
- Modelica.Electrical.Digital: {127,0,127}
- Modelica.Electrical.Machines: {0,0,255}
- Modelica.Electrical.MultiPhase: {0,0,255}
- Modelica.Electrical.QuasiStationary: {85,170,255}
- Modelica.Electrical.Spice3: {170,85,255}
- Modelica.Magnetic.FluxTubes: {255,127,0}
- Modelica.Magnetic.FundamentalWave: {255,128,0}
- Modelica.Magnetic.QuasiStatic: {255,170,85}
- Modelica.Mechanics.MultiBody: {192,192,192}
- Modelica.Mechanics.Rotational: {95,95,95}
- Modelica.Mechanics.Translational: {0,127,0}
- Modelica.Fluid: {0,127,255}
- Modelica.Media: none
- Modelica.Thermal.FluidHeatFlow: {0,0,255}
- Modelica.Thermal.HeatTransfer: {191,0,0}
- Modelica.Math: none
- Modelica.ComplexMath: none
- Modelica.Utilities: none
- Modelica.Constants: none
- Modelica.Icons: none
- Modelica.SIunits: none

I suggest to add this list to the icon style guide as well.

christiankral, 2018-05-16:
@tobolar Thanks for providing the color code details.

I will be adding the color codes to the User's Guide. However, there are two block exceptions to the rule that shall be discussed, as these blocks do NOT use the standard block color. 

### 1. Boolean blocks ###

- Modelica.Blocks.Logical
- Modelica.Blocks.MathBoolean

There is an icon Modelica.Blocks.Icons.BooleanBlock defined, but apparently not used. Instead the discrete block icons are used for all boolean blocks. Yes, boolean variables are discrete variables, but even the block frame is not magneta... I also understand, that there are color discrepancies with blocks which have real inputs and boolean outputs, etc.

### 2. Integer blocks ###

- Modelica.Blocks.MathInteger

Here the icon Modelica.Blocks.Icons.BooleanBlock is defined but not used. Instead an orange filled block icon is used. I suppose the filled block shall again suggest that these are discrete blocks.

dietmarw, 2018-05-16:
> 1. ...
> There is an icon Modelica.Blocks.Icons.BooleanBlock defined, but apparently not used. Instead the discrete block icons are used for all boolean blocks. 

Are you sure. What I can see is that they boolean blocks extends from `Modelica.Blocks.Icons.PartialBooleanBlock` and I can not find a reference to discrete there. Could you point to a specific example where this is the case?

> 2. ...
> Here the icon Modelica.Blocks.Icons.BooleanBlock is defined but not used. 
Copy paste error?

HansOlsson, 2018-05-16:
Regarding BooleanBlock - the following blocks seem to use it:
Modelica.Blocks.Interfaces.BooleanMIMOs Modelica.Blocks.Routing.BooleanReplicator Modelica.Blocks.Interfaces.BooleanSignalSource Modelica.Blocks.Interfaces.BooleanSISO Modelica.Blocks.Interfaces.IntegerMIBooleanMOs Modelica.Blocks.Interfaces.IntegerSIBooleanSO Modelica.Blocks.Interfaces.MI2BooleanMOs Modelica.Blocks.Interfaces.SI2BooleanSO

However, I don't understand the documentation of "PartialBooleanBlock". 

It seems almost as though it is a special icon for partial blocks. That does not make sense, since partial blocks are usually extended and become non-partial (as in the Logical Library).
I don't think that is just me misinterpreting the text since the look of "PartialBooleanBlock" is similar to the look for a replaceable block - which makes a replaceable Modelica.Blocks.Interfaces.partialBooleanSISO look like it is twice replaceable.

tobolar, 2018-05-16:
Please notice the issue #2512 which also touch Modelica.Blocks.Logical and Modelica.Blocks.MathBoolean.

dietmarw, 2018-05-16:
I also notice that lots of blocks in `Modelica.Icons.InterfacesPackage` have the wrong capitalisation of partial. They should start with capital P. Probably should become a ticket for the next major release.

beutlich, 2018-05-16:
Getting OT: We discussed somewhere (cannot find it) that having `Partial` as class name prefix does not seem to be a good idea. For that reason I created partial base classes with the `Base` suffix (e.g., Modelica.Blocks.Tables.Internal.CombiTable2DBase or Modelica.Mechanics.MultiBody.Interfaces.LineForceBase).

harmanpa, 2018-05-16:
@dietmarw You mean `Modelica.Blocks.Interfaces`, see #609

tobolar, 2018-05-16:
I would suggest to collect all the points and to resolve it at once for the next major version.

christiankral, 2018-05-16:
Added color codes and samples to Modelica.UsersGuide.Conventions.Icons in 36d9e4b3098fc97c37cf85c59aa9fdb722933bb7.

christiankral, 2018-05-16:
So how do we proceed with a possible change of boolean and integer blocks? 
On my opinion we shall close this issue and create a new ticket to re-design boolean and integer blocks. Personally I have no problems to go for a different icon look in MSL 3.2.3. However, I currently do not have the resources to make a proposal to be included in the MSL 3.2.3. So either someone else can make a proposal or we shift this issue to the next MINOR (or MAJOR) release.

dietmarw, 2018-05-16:
I think there is no real reason for a rush. MSL can always become tidier and nicer to look at. So I'd say postponing this until the next major release is a good idea if this means also being able to clean up the naming.

christiankral, 2019-02-10:
I think it may make sense to unify the appearance of Modelica blocks. 

Comparing some different models and blocks shows:

[image]

The icons of components have the same color as specified in the "domain color", see `Modelica.UsersGuide.Conventions.Icons`. The different blocks, however, have a different design. The lines representing the internal behavior of the blocks should also follow a unified design scheme. To me it makes sense to use 
- the domain color (in most cases)
- gray for axes and curves
- black (or dark gray) for symbols and labels; maybe similar to #2628 and #2627

christiankral, 2019-06-16:
Unified colors of ComplexBlocks are discussed in #2833 and handled in #2977.

christiankral, 2019-06-16:
I guess the original intention of this ticket is covered by now. 

The "only" missing thing is the improvement of the icons of `Modelica.Blocks`. Probably we should create a new ticket based on the conclusions of this ticket:
 
- `Modelica.Blocks.Icons.PartialBooleanBlock` shall be renamed; however, I do not understand what the difference of `Blocks.Icons.PartialBooleanBlock` and `Blocks.Icons.BooleanBlock` in terms of applications is. When shall I use which icon? If I had the choice I would go for `Blocks.Icons.BooleanBlock` as it is more in line with all the other block designs.
- `Modelica.Blocks.Interfaces.partialBooleanSISO` and similar partial blocks shall start with an uppercase P; however, the name may be changed anyways due to #609
- What is the difference between `Modelica.Blocks.Interfaces.BooleanSISO` and `Modelica.Blocks.Interfaces.partialBooleanSISO`: Just a different Icon? Shouldn't the names somehow be re-structured and unified?
- Change fill color of `Modelica.Blocks.Interfaces.partialBooleanSISO` and `Modelica.Blocks.Interfaces.PartialIntegerMISO` or is there a particular reason I cannot understand?
- Questionanle `initScale`, see #2512

I prefer closing this ticket now. Other opinions?

beutlich, 2019-11-02:
> I prefer closing this ticket now.

OK. Thanks.

closed 2019-11-02 by beutlich