modelica/ModelicaStandardLibrary #2479: Partial interfaces in particular package in Modelica.Thermal.FluidHeatFlow.Interfaces

tobolar, 2018-02-28:
There are collected partial interface classes in a particular subpackage `Modelica.Thermal.FluidHeatFlow.Interfaces.Partials`. This is not common in the MSL and the classes should be moved one level higher, i.e. into `Modelica.Thermal.FluidHeatFlow.Interfaces`.

christiankral, 2018-03-05:
We also use sub packages in Modelica.Electrical.Machines.Interfaces to structure the partial models. I think it would even make sense to structure Modelica.Electrical.Machines.Interfaces by the data types boolean, integer and real.

tobolar, 2018-03-06:
In Modelica.Electrical.Machines.Interfaces, the situation maybe different to FluidHeatFlow.Interfaces.
I would rather compare to
- `Modelica.Fluid.Interfaces`,
- `Modelica.Mechanics.Rotational.Interfaces` or
- `Modelica.Thermal.HeatTransfer.Interfaces`.

The interfaces defined in those packages reveal a similar idea, but the Partials subpackage is used only in FluidHeatFlow.

christiankral, 2019-06-22:
I agree, we shall move the partials one level up and keep the names as they are. I will prepare a PR.

PR #2997 (2019-07-17): Move FluidHeatFlow models of Interfaces.Partials to Interfaces
close #2479
closed 2019-07-17 by AHaumer