modelica/ModelicaStandardLibrary #2482: MultiBody models have a problem with StateSelect.always/never

sjoelund, 2018-03-05:
When trying to implement a check [in OpenModelica](https://github.com/OpenModelica/OMCompiler/pull/2191) for variables chosen as states despite StateSelect.never and variables not selected as states despite having StateSelect.always, I discovered that Joints.Revolute and Parts.RollingWheel have this problem. For both models, the problem is that you set these models using a single stateSelect parameter.

For Joints.Revolute, the stateSelect parameter is used for both phi and w where `w = der(phi)` and `a = der(w)` which means these might reasonably be chosen as states in some models (depending on optimizations); so StateSelect.never is probably not a good value. Modelica.Mechanics.MultiBody.Examples.Constraints.RevoluteConstraint does set StateSelect.never and the question is if there should be an assertion checking that StateSelect.never was used, if this should trigger a warning in the tool and the example be changed, or if the model should simply use `SI.Angle phi(start=0, final stateSelect=max(stateSelect, StateSelect.avoid))`.

For Parts.RollingWheel we have the opposite problem: at least OpenModelica gets for Elementary.RollingWheel `wheel1.der_angles[1]` and `wheel1.der_angles[2]` as non-state variables with stateSelect=StateSelect.always (regardless of index reduction method used). And it stands to reason that if the acceleration of these variables is not used, StateSelect.always is a bad value; but it might be fine for the other variables in the model (and indeed setting StateSelect.prefer in Elementary.RollingWheel is problematic for OpenModelica). So again the question is if we should guard der_angles in Parts.mo with `each stateSelect=min(StateSelect.prefer, stateSelect)` or if we should fix the RollingWheel example by for example setting stateSelect=StateSelect.always and manually overriding only `der_angles` (but this seems like an awkward thing to do).

tobolar, 2018-03-05:
See also #1810.

HansOlsson, 2018-04-16:
> Modelica.Mechanics.MultiBody.Examples.Constraints.RevoluteConstraint does set StateSelect.never and the question

I agree that there is a problem here, and will discuss that in #1810

> For Parts.RollingWheel we have the opposite problem: at least OpenModelica gets for Elementary.RollingWheel wheel1.der_angles[1] and wheel1.der_angles[2] as non-state variables with stateSelect=StateSelect.always (regardless of index reduction method used). 

I cannot see any problem with this model in Dymola, where wheel1.der_angles become state-variables.