modelica/ModelicaStandardLibrary #2485: Rename parameters of Modelica.Mechanics.MultiBody.Visualizers.Plane

beutlich, 2018-03-06:
Parameters of Modelica.Mechanics.MultiBody.Visualizers.Plane are labeled axis_[xy], length_[xy] and n[xy]. Setting axis_x to {0,1,0} and axis_y to {1,0,0} might cause some confusion. I rather would like to use the same naming convention as Modelica.Mechanics.MultiBody.Visualizers.FixedShape, e.g., rename
* axis_x -> lengthDirection
* axis_y -> widthDirection
* length_x -> length
* length_y -> width

For the discretization parameters I am not that sure and simply propose
* nx -> n
* ny -> m

See #2424.

tobolar, 2018-03-06:
You are right.
For surfaces, there are generally used directions u and v. So what about to simply change
- [xy] -> [uv]?

Then:
- axis_x -> direction_u
- axis_y -> direction_v
- length_x -> length_u
- length_y -> width_v
- nx -> nu
- ny -> nv

MartinOtter, 2018-03-06:
Please do not make such changes without larger discussion.
It was a mistake to introduce "length", "width" etc. in the MultiBody library, because it is not clear whether this is a distance in x,y, or z direction. It is much clearer, if "x", "y", "z" are appended in the name, because it is then completely clear, without reading the documentation, which coordinate axis is meant. 

Therefore, if you make changes, do it the other way round, so replace lengthDirection by direction_x or something like that.

beutlich, 2018-03-06:
I feel more comfortable with the names I proposed above. But let us discuss it in larger group.

beutlich, 2018-03-06:
> t was a mistake to introduce "length", "width" etc. in the MultiBody library

If you consider it now as a mistake, we really should have a GitHub issue for such kind of issues. Otherwise, we will never find out or remember.

beutlich, 2019-07-13:
@tobolar I set the milestone to MSL4.0.0 in case we want to do some renaming. Can you please propose your changes fixing the "mistakes" in the MultiBody lib?