modelica/ModelicaStandardLibrary #2486: Use dark green color in icons of Modelica.Mechanics.Translational

tobolar, 2018-03-06:
The icon style guide Modelica.UsersGuide.Conventions.Icons states that "The main icon color of a component shall be the same for all components of one library." The library `Modelica.Mechanics.Translational` mostly uses black color ({0,0,0}), but deducing from interfaces defined (e.g. `Modelica.Mechanics.Translational.Interfaces.Flange_a`), dark green color ({0,127,0}) should be used instead.

Figure below shows how the changes should look like.
[image]

beutlich, 2018-03-06:
As a remark: We previously had the discussion that changing the visual appearance of icons might feel like a backwards incompatible change.

beutlich, 2018-04-24:
@tobolar To answer your question from https://github.com/modelica/ModelicaStandardLibrary/issues/2186#issuecomment-383599945: It is up to the library officer to decide if you want to change the icons for MSL v3.2.3 or the next major release. As you are a library officer (together with @AHaumer, @MartinOtter and @christiankral) you should come to a mutual agreement on the milestone.

tobolar, 2018-04-24:
@beutlich Let me cite your comment again: 
> ... we previously had the discussion that changing the visual appearance of icons might feel like a backwards incompatible change

Thus I have got the impression that this is a general remark on MSL beeing respected for all sub-libraries.

beutlich, 2018-04-24:
Sorry for the confusion. I only learned myself.

Feel free to decide with your fellow library officers to target the milestone.

tobolar, 2018-04-24:
@AHaumer, @MartinOtter, @christiankral What is your opinion regarding changing the icons' color in the next MINOR version, i.e. 3.2.3?

beutlich, 2018-04-24:
Note, that in https://github.com/modelica/ModelicaStandardLibrary/issues/2215#issuecomment-289205204 @christiankral stated that he changed the icon colors of Electrical.Analog and for QuasiStationary components. Thus, if a color change was acceptable for one sublibrary it should be treated similar in other sub-libraries.

christiankral, 2018-04-24:
I am very much in favor of everything that improves the consistency and coherence of the MSL. I therefore also support the change of the icon colors.

christiankral, 2018-04-25:
To be more precise: On my opinion we should not wait for the next major release when improving consistency and coherence of the MSL.

GallLeo, 2018-04-25:
I'm also in favor of adapting the apperance in minor releases (probably better not in patches).
That's in line with the last bigger change in appearance which happend from MSL 3.2 to MSL 3.2.1. 
Back then, Wolfram proposed many changes and some of them made it into MSL 3.2.1. See #1096

One guideline from back then was to _not_ change colours of connectors, because we do not have conversion for them. But, changing icons should be fine. Potentially, you could visually "break" user models, if they extend from MSL components. 

This ticket doesn't want to change connectors, so it should be fine.

MartinOtter, 2018-04-26:
I have no strong opinion. Just go on.

tobolar, 2018-04-26:
Rather positive reactions. So I will change the icons.

beutlich, 2018-04-26:
Should the damper fill color (form you screen shot above) also be greenish?

tobolar, 2018-04-26:
Well ... I rather tend to not use that green color in the damper. The rod, in contrast, should look similar to translational mass, and should be of green color or a shade of it.

tobolar, 2018-04-27:
@beutlich :smiley: 
Please let me prepare the first draft (PR) we can discuss then, ok?

closed 2018-05-06 by beutlich