modelica/ModelicaStandardLibrary #2488: Add unit and displayUnit to signal input and output representing an angle

christiankral, 2018-03-13:
There are a couple models in the MSL which provide an angle signal input or output. However, when simulating and displaying such angles, the displayUnit can often NOT be switched to "deg", since the output is just a RealOutput with no unit. Therefore, the angle signal input and output shall be equipped with ```(unit="rad",displayUnit="deg")```.

This issue applies to the following models (list may not yet be complete):
- Modelica.Blocks.Math.Sin
- ...
- Modelica.Blocks.Math.Atan2
- Modelica.Blocks.Math.WrapAngle
- Modelica.Blocks.Math.RectangularToPolar
- Modelica.Blocks.Math.RectangularToPolar
- Modelica.Blocks.Math.Harmonic 
- Modelica.ComplexBlocks.ComplexMath.PolarToComplex
- Modelica.ComplexBlocks.ComplexMath.ComplexToPolar
- Modelica.ComplexBlocks.ComplexMath.Bode
- Modelica.Electrical.Machines.SpacePhasors.Components.Rotator
- Modelica.Electrical.Machines.SpacePhasors.Blocks.Rotator
- Modelica.Electrical.Machines.SpacePhasors.Blocks.ToPolar
- Modelica.Electrical.Machines.SpacePhasors.Blocks.FromPolar
- Modelica.Electrical.Machines.Utilities.ToDQ
- Modelica.Electrical.Machines.Utilities.FromDQ
- Modelica.Magnetic.QuasiStatic.FundamentalWave.Utilities.CurrentController
- Modelica.Electrical.Machines.Utilities.CurrentController
- Modelica.Electrical.Machines.Utilities.VoltageController

HansOlsson, 2018-03-13:
> There are a couple models in the MSL which provide an angle signal input or output. However, when simulating and displaying such angles, the displayUnit can often NOT be switched to "deg", since the [...]

I can understand the need for unit="rad", and I would also hope that we could derive that unit in more cases. However, it may be excessive to use displayUnit="deg" - when looking at polar representation I don't necessarily consider degrees a better choice.

AHaumer, 2018-03-18:
see #2494 (and #2492), let's continue the discussion on a generalized level in #2494.

christiankral, 2018-05-16:
OK, to me this sounds like adding the angle unit "rad" is possible without getting the MSL backwards incompatible, right? The displayUnit shall not be included -- so this makes sense to me. I can create a pull request, if we agree to go for adding units to angle inputs and outputs.

AHaumer, 2018-05-20:
I agree with Hans, displayUnit might be overdone.
It seems that unit (without diplayUnit) won't cause problems or incompatiblities. 
I suggest starting implementation, but I think we won't get all models done for 3.2.3 
Therefore shifting the milestone.

AHaumer, 2018-05-21:
But don't forget angular velocity ;-)

AHaumer, 2018-05-21:
I'm just working on the initially mentioned.
For two blocks this isn't possible:
- Modelica.Electrical.Machines.SpacePhasors.Blocks.ToPolar
- Modelica.Electrical.Machines.SpacePhasors.
since the output is a vector of lenght 2, the first element representing the length and the second the angle. Don't know how to set the unit of just 1 element of an arry.

To be sure: We should go through the whole standard library a look for similar occurenes.
Blocks.FromPola

AHaumer, 2018-05-21:
To be sure: We should go through the whole standard library a look for similar occurenes.

commit 204545c6 (2018-05-21): partly resolves #2488
commit 8101fdec (2018-05-21): partly resolves #2488
PR #3267 (2020-01-02): Add units to real I/O signals
Though decided in #2492 to utilize `final unit="X"`, I discared the `final`. For me, the unit of real I/O signals currently seems arbitrary and uncontrolled in MSL. Some use, `unit="X"`, some `final unit="X"`, some with `displayUnit` and some even with `quantity`. There could be anther issue to harmonize these kind of unit modifications.

@christiankral @HansOlsson @AHaumer Please check carefully when reviewing. Thanks.

Closes #2488, closes #2492.
closed 2020-01-02 by beutlich