modelica/ModelicaStandardLibrary #2493: Modelica.Electrical.Analog.Ideal.AD_Converter internal error check fails w/ OpenModelica (v1.11.0 Win64)

jpaximadis, 2018-03-17:
Not sure if this is a Modelica standard library issue or an OpenModelica issue, but it was resolved by...

1) Adding:
> import SI = Modelica.SIunits;

2) Changing:

> initial equation
>   for i in 1:N loop
>     y[i] = L.'X';
>   end for;
> 

To:

> initial algorithm
>   for i in 1:N loop
>     y[i] := L.'X';
>   end for;

beutlich, 2018-05-02:
@sjoelund Is this a tool issue?

sjoelund, 2018-05-02:
Is this used in any example? The error only seems to trigger for `checkModel` functionality (which is rarely used except for models that are incomplete such as this one).

beutlich, 2018-05-02:
> Is this used in any example?

Yes, in Modelica.Electrical.Analog.Examples.AD_DA_conversion

sjoelund, 2018-05-02:
And that model works just fine. So there is only a problem trying to check an incomplete model that has a for-loop such as this one; so it's a minor tool issue.

beutlich, 2018-05-02:
Well, we can change it as proposed to make OMC happy.

commit 6b76c2f2 (2018-05-07): close #2493: Move initialization to algorithm
closed 2018-05-07 by beutlich