modelica/ModelicaStandardLibrary #2494: Add quantities and units to signal inputs and outputs

AHaumer, 2018-03-18:
This issue generalizes #2488 and #2492:
IMHO it is a good idea to add both quantities and units to signal inputs and outputs whenever they're unambiguous, e.g. a sensor output or the input of a source. 
In some places of the MSL these modiifiers are already implemented. 
This would make it easier to show a signal using a convenient unit (e.g. Angle, AngularVelocity, Temperature, ...) choosen by the user even when a tool is not able to deduce the unit automatically. 
Of course there are lots of blocks with signal inputs and outputs where we're not able to specifiy the units, e.g. a FirstOrder. They may be used for many different signals with different units.  
Additionally, there are blocks whith such unspecific (regarding quantity and unit) inputs and outputs, but with a distinct relationship between the units of input and output (and parameters), e.g. an Integrator: if the actual input has the unit "u" (maybe deduced by a tool), the output has to have the unit "u.s" due to integration. 
Any comments to this issue? Are there pitfalls? Who would support the task of inspecting the whole MSL and adding the modfiers (quantities and units)?

beutlich, 2018-03-19:
See also #830, #872, #876 and especially #1125, where we discussed that adding quantitiy attributes is a non-backward compatible change with unspecified semantics.

HansOlsson, 2018-04-13:
> Any comments to this issue? Are there pitfalls?

Apart from the already mentioned issues, it still seems unclear what the "quantity" really means and if we use the standard names.  

As an example angles aren't called "angle" but "plane angle" in the SI-brochure; similarly resistance is called "electrical resistance" in the SI-brochure.  And consider these ones:

```
quantity:(energy, work, amount of heat): joule J N m m2 kg s−2
  type Work = Real (final quantity="Work", final unit="J");
  type Energy = Real (final quantity="Energy", final unit="J");
quantity:(electric charge, amount of electricity), coulomb C s A
  type ElectricCharge = Real (final quantity="ElectricCharge", final unit="C");
quantity:(electric potential difference, electromotive force), volt V W/A m2 kg s−3 A−1
  type ElectricPotential = Real (final quantity="ElectricPotential", final unit=
         "V");
  type Voltage = ElectricPotential;
  type PotentialDifference = ElectricPotential;
  type ElectromotiveForce = ElectricPotential;
```

HansOlsson, 2018-04-13:
For units there is less ambiguity than for quantity, but on the other hand tools can uniquely derive them - so there seems less reason to add them.

The word "uniquely" makes this different from initialization where we previously spent time to set fixed=true for specific start-values, even if tools could deduce initial conditions.

Specifically for Blocks there is an issue that people might use them for signals without any physical unit; and unit-checking can be more of a problem than a help in those cases - e.g. you just write `der(x)=-x;` - graphically using blocks it seems unlikely that a unit-warning for der(x) and x having different unit will help you.

beutlich, 2018-04-19:
See also c641a86c29b5ff326b99e14525b45f08b9f3f18a for #2000.

beutlich, 2018-04-20:
Since the quantity attribute is considered ambiguous, the focus is on the unit attribut, where we have the original issues #2488 and #2492 to deal with.

closed 2018-04-20 by beutlich