modelica/ModelicaStandardLibrary #2498: OMC links in Release Notes

beutlich, 2018-04-11:
The OMC related links to Compilation, Simulation or Regressing testing in [Modelica.UsersGuide.ReleaseNotes.Version_3_2_2](http://doc.modelica.org/om/Modelica.UsersGuide.ReleaseNotes.Version_3_2_2.html) do not work or are not reproducible w.r.t MSL v3.2.2 and OpenModelica 1.9.4 Beta.2.

sjoelund, 2018-04-11:
We have a new library testing script where we can easier compare the MSL branches to the different OMC branches: https://libraries.openmodelica.org/branches/overview.html
I would suggest archiving it somewhere if it's linked in the release notes though.

beutlich, 2018-04-11:
> I would suggest archiving it somewhere if it's linked in the release notes though.

Yes, that seems to be useful.

Any chance to reproduce the MSL v3.2.2 results? If not, I'd recommend to remove these links.

sjoelund, 2018-04-11:
Well, we could possibly reproduce part of the results by running the old OMC version with the old library testing script. But the links to Hudson would be dead. This report would be pretty close though (the 1.9.7 is the maintenance release of 1.9.4; unlike 1.9.3 which is a totally separate release): https://libraries.openmodelica.org/branches/v1.9/Modelica_3.2.2/Modelica_3.2.2.html. But it's running on the 3.2.2 maintenance branch.