modelica/ModelicaStandardLibrary #2512: Questionable initialScale in some Blocks

tobolar, 2018-04-25:
In the following blocks, there is used initialScale=0.06 without obvious benefit.
- Modelica.Blocks.Interfaces.PartialBooleanMISO
- Modelica.Blocks.Interfaces.PartialIntegerMISO
- Modelica.Blocks.Interfaces.PartialIntegerSISO
- Modelica.Blocks.Interfaces.PartialRealMISO
- Modelica.Blocks.Sources.RadioButtonSource
- Modelica.Blocks.MathInteger.TriggeredAdd

If there is a benefit, the same scaling should be used consequently within other (partial) classes such as Modelica.Blocks.Interfaces.SISO.

In contrast, in Modelica.Blocks.Interfaces.PartialBooleanSISO_small, the name at least indicates the scaling. Though it may not be obvious to the user what feature is reflected by "small".
See also #2419.

MartinOtter, 2018-04-26:
Please look at the blocks where these interfaces are used (like MultiSum). The purpose was to have "smaller" blocks for "trivial" operations like "*" or "+". Just try it out by dragging the corresponding blocks.

tobolar, 2018-04-26:
I understand that it is reasonable to have smaller blocks in some cases. But my impression is that this decission is fully arbitrary.

Regarding only **MISO** partial blocks, there is set initialScale=0.06 in 
* Modelica.Blocks.Interfaces.PartialBooleanMISO
* Modelica.Blocks.Interfaces.PartialIntegerMISO
* Modelica.Blocks.Interfaces.PartialRealMISO

but no such definition in (complementary?)
* Modelica.Blocks.Interfaces.MISO

The situation is even worse for **SISO**. initialScale=0.06 is set only in
* Modelica.Blocks.Interfaces.PartialIntegerSISO

but no such definition e. g. in
* Modelica.Blocks.Interfaces.DiscreteSISO
* Modelica.Blocks.Interfaces.BooleanSISO
* Modelica.Blocks.Interfaces.partialBooleanSISO

And there are far more partial blocks MIMO, SIMO.

But maybe better to consider the issue from **another perspective**. We can compare the blocks package-wise, e.g. from  Logical and MathBoolean since the blocks in those packages often look very similar. Then:
* `Modelica.Blocks.Logical`: Blocks extend mostly from `partialBooleanSISO` and have "standard" size.
* `Modelica.Blocks.MathBoolean`: Block extends from `PartialBooleanMISO` or `PartialBooleanSISO_small` and are scaled.

So my question is actually _how is the decission procedure_ whether (and how) the blocks are scaled per default or not. Once such the procedure is clear, it should be documented in 
`Modelica.UsersGuide.Conventions.Icons`.

tobolar, 2021-01-21:
Resolve  #2164 first.