modelica/ModelicaStandardLibrary #2534: Replace Modelica.Mechanics.MultiBody.Interfaces.PartialAbsoluteSensor

HansOlsson, 2018-05-18:
Based on comment in #609 I would suggest to replace Modelica.Mechanics.MultiBody.Interfaces.PartialAbsoluteSensor by Modelica.Mechanics.MultiBody.Sensors.Internal.PartialAbsoluteSensor

The reasons are:

-

beutlich, 2018-10-19:
@HansOlsson How to proceed exactly for the next major release?

1. Move Modelica.Mechanics.MultiBody.Interfaces.PartialAbsoluteSensor to some Obselete-library
1. Move Modelica.Mechanics.MultiBody.Sensors.Internal.PartialAbsoluteSensor to Modelica.Mechanics.MultiBody.Interfaces.PartialAbsoluteSensor
1. Add conversion script to rename class Modelica.Mechanics.MultiBody.Interfaces.PartialAbsoluteSensor to ObsoleteLib.SomePackage.PartialAbsoluteSensor 
1. Add conversion script to rename class Modelica.Mechanics.MultiBody.Sensors.Internal.PartialAbsoluteSensor to Modelica.Mechanics.MultiBody.Interfaces.PartialAbsoluteSensor

Is the safe? I mean running the conversion script twice will rename the class again. Is it therefore better to introduce a different class name PartialAbsoluteSensor2 in Interfaces?

And what about Modelica.Mechanics.MultiBody.Interfaces.PartialRelativeSensor? This class is also neither uitilized in Modelica nor ModelicaTest.

tobolar, 2018-10-19:
I have checked #609 again and there is still the (unfortunately undecided) question on the usage of `BasesPackage` package collecting base classes. A possible solution would be then

2. Move Modelica.Mechanics.MultiBody.Sensors.Internal.PartialAbsoluteSensor to Modelica.Mechanics.MultiBody.BaseClasses.AbsoluteSensor (similarly named as Modelica.Electrical.Analog.Interfaces.AbsoluteSensor)

I personally would prefer such the solution. (I mean to distinguish between Interfaces and BaseClasses)

beutlich, 2019-06-21:
> I personally would prefer such the solution. (I mean to distinguish between Interfaces and BaseClasses)

@tobolar I suggest you go ahead with the clean separation of BaseClasses and Unterfaces for MultiBody.

tobolar, 2019-07-09:
@beutlich 
Unfortunately, not so far - see https://github.com/modelica/ModelicaStandardLibrary/pull/2997#issuecomment-509642149