modelica/ModelicaStandardLibrary #2554: FluidHeatFlow: Enhancement of piston model and pump model

AHaumer, 2018-05-28:
Maybe this can be fixed earlier (MSL 3.2.3):

The **piston model** should consider the mass of the cylinder (in case useSupport=true) and the piston.

The **pump model** should get an optional support, 
and it should consider the inertia of the housing (in case useSupport=true) and the punp wheel.
The calculation of mechanical power resp. torque should be improved.
Documentation should be improved (description of the pump characteristic, the pump can act as a turbine).

AHaumer, 2018-05-28:
0ade4359bd76c44cfd6d20e3102f606fd6283708:
Implements the optional support of the pump model, and improved its documentation

beutlich, 2019-12-23:
@AHaumer The cylinder mass is still to do, right?