modelica/ModelicaStandardLibrary #2556: Redesign FreeMotionScalarInit to not influence dynamic simulation when no states are selected

tobolar, 2018-05-29:
In #1810, @HansOlsson discuss the problematic design of `Modelica.Mechanics.MultiBody.Joints.FreeMotionScalarInit` concerning its influence on dynamic simulation. 
A solution should be found and implemented on calculation of angles, angular velocity and acceleration when no states are selected. The suggested solution is
> replacing its internal InitAngle (and its derivative) by the corresponding sensor components

See also PR #2529.

beutlich, 2019-09-05:
@tobolar This is labeled "high" and orginated from https://github.com/modelica/ModelicaStandardLibrary/pull/2529#issuecomment-392700692. Do you think you can provide a fix for MSL v4.0.0?

tobolar, 2019-11-28:
I will look at it again.

beutlich, 2019-11-28:
Thanks. Just want to make sure that "high" issues are not forgotten.

tobolar, 2020-02-11:
A possible solution is prepared in https://github.com/tobolar/ModelicaStandardLibrary/tree/issue2556_FreeMotionScalarInit. Will wait with a PR for merging of #3396.

beutlich, 2020-02-11:
> Will wait with a PR for merging of #3396.

Done.