modelica/ModelicaStandardLibrary #2560: Change units of potential quanities to potential units

christiankral, 2018-05-29:
The electrical and magnetic potentials of some connector definition use a potential difference in the connector definitions. This seems 
1. awkward, since it the potential quantities shall be defined using a potential definition,
2. not necessary, since potential and potential difference definitions are identical

The change is fully backwards compatible and shall be applied to the upcoming MSL 3.2.3.

# Modelica.Electrical.Analog

```modelica
connector Pin "Pin of an electrical component"
  SI.Voltage v "Potential at the pin"; 
end Pin;
``` 
shall be changed to:

```modelica
connector Pin "Pin of an electrical component"
  SI.ElectricPotential v "Potential at the pin"; 
end Pin;
``` 

justified by the following defintions of Modelica.SIunits:

```modelica
  type ElectricPotential = Real (final quantity="ElectricPotential", final unit="V");
  type Voltage = ElectricPotential;
  type PotentialDifference = ElectricPotential;
```

# Modelica.Electrical.QuasiStationary.SinglePhase

```modelica
connector Pin "Basic connector"
  Modelica.SIunits.ComplexVoltage v "Complex potential at the node";
  flow Modelica.SIunits.ComplexCurrent i "Complex current flowing into the pin";
end Pin;
```

shall be changed to:

```modelica
  connector Pin "Basic connector"
  Modelica.SIunits.ComplexPotential v "Complex potential at the node";
  flow Modelica.SIunits.ComplexCurrent i "Complex current flowing into the pin";
end Pin;
```

justified by the following defintions of Modelica.SIunits:

```modelica
  operator record ComplexElectricPotential =
    Complex(redeclare Modelica.SIunits.ElectricPotential re "Imaginary part of complex electric potential",
            redeclare Modelica.SIunits.ElectricPotential im "Real part of complex electrical potential")
    "Complex electric potential";
  operator record ComplexPotentialDifference =
    Complex(redeclare Modelica.SIunits.PotentialDifference re "Real part of complex potential difference",
            redeclare Modelica.SIunits.PotentialDifference im "Imaginary part of complex potential difference")
    "Complex electric potential difference";
  operator record ComplexVoltage =
    Complex(redeclare Modelica.SIunits.Voltage re "Imaginary part of complex voltage",
            redeclare Modelica.SIunits.Voltage im "Real part of complex voltage")
    "Complex electrical voltage";
```

# Modelica.Magnetic.FundamentalWave

```modelica
connector MagneticPort "Complex magnetic port"
  Modelica.SIunits.ComplexMagneticPotentialDifference V_m
    "Complex magnetic potential difference";
  flow Modelica.SIunits.ComplexMagneticFlux Phi "Complex magnetic flux";
end MagneticPort;
```

shall be changed to:

```modelica
connector MagneticPort "Complex magnetic port"
  Modelica.SIunits.ComplexMagneticPotential V_m
    "Complex magnetic potential difference";
  flow Modelica.SIunits.ComplexMagneticFlux Phi "Complex magnetic flux";
end MagneticPort;
```

justified by the following defintions of Modelica.SIunits:

```modelica
  operator record ComplexMagneticPotential =
    Complex(redeclare Modelica.SIunits.MagneticPotential re "Real part of complex magnetic potential",
            redeclare Modelica.SIunits.MagneticPotential im "Imaginary part of complex magnetic potential")
    "Complex magnetic potential";
  operator record ComplexMagneticPotentialDifference =
    Complex(redeclare Modelica.SIunits.MagneticPotentialDifference re "Real part of complex magnetic potential difference",
            redeclare Modelica.SIunits.MagneticPotentialDifference im "Imaginary part of complex magnetic potential difference")
    "Complex magnetic potential difference";
```

and 

```modelica
  type MagneticPotential = Real (final quantity="MagneticPotential", final unit="A");
  type MagneticPotentialDifference = Real (final quantity=
          "MagneticPotential", final unit="A");
```

closed 2018-05-29 by beutlich

christiankral, 2018-06-05:
What is the better wording of a connector definition?

1. `Potential at the pin` (currently used)
2. `Potential of pin` 
3. `Potential of the pin` (better grammar, possibly)

I think it makes sense to unify the wording of the electrical and magnetic connectors -- as they appear all so different...

dietmarw, 2018-06-05:
My vote would be "3.".

harmanpa, 2018-06-05:
My vote would be "1"

HansOlsson, 2018-06-05:
I tried to look and "potential of the" can risk confusion with another use of this - e.g. "potential of the field generated by a point source" (which is quite different).

"potential at the pin" avoid that problem (proposal 1).

dietmarw, 2018-06-05:
...but "1." would be also fine for me. :-)

christiankral, 2018-06-05:
Thanks, I'll go for 1. :+1:

christiankral, 2018-06-05:
What is the correct spelling of quasi-static?

1. quasistatic
2. quasi static
3. quasi-static (provided by http://www.electropedia.org/iev/iev.nsf/display?openform&ievref=113-04-08)

@harmanpa you please help here?

harmanpa, 2018-06-05:
I've done a quick poll of those sat near me and we say "quasi-static"

christiankral, 2018-06-05:
Thanks a lot. I will use quasi-static then.

commit b614605c (2018-06-13): fixes #2560 (usage of obsolete pipe model) (#2625)