modelica/ModelicaStandardLibrary #2574: Potentially illegal conditional components in Fluid.Vessels

perost, 2018-06-01:
`Modelica.Fluid.Vessels.BaseClasses.PartialLumpedVessel` contains the following conditional components (type names shortened for readability):
```Modelica
RealInput[nPorts] portsData_diameter_internal = portsData.diameter if use_portsData and nPorts > 0;
RealInput[nPorts] portsData_height_internal = portsData.height if use_portsData and nPorts > 0;
RealInput[nPorts] portsData_zeta_in_internal = portsData.zeta_in if use_portsData and nPorts > 0;
RealInput[nPorts] portsData_zeta_out_internal = portsData.zeta_out if use_portsData and nPorts > 0;
```
`portsData` is in turn declared as:
```Modelica
parameter VesselPortsData[if use_portsData then nPorts else 0] portsData;
```
The issue with this is that when `use_portsData = false` then `portsData` will be an array with 0 elements while the conditional variables above will have `nPorts` elements, leading to a type mismatch between the variables and their binding equations when `nPorts > 0`. This occurs in e.g. `Modelica.Fluid.Examples.TraceSubstrances.RoomCO2WithControls` where `use_portsData` is set to `false` and `nPorts` to 4 on the `volume` component.

This was previously somewhat of a gray area since it wasn't really specified how valid conditional components should be, but in Modelica 3.4 this was clarified as (section 4.4.5):
>If the Boolean expression is false the component (including its modifier) is removed from the flattened DAE , and connections to/from the component are removed. [Adding the component and then removing it ensures that the component is valid.]

I interpret this to mean that conditional components must be as valid as any other component, including having type compatible binding equations.

perost, 2018-06-01:
I guess a simple fix could be to just use the declare the conditional components to have the same dimensions as `portsData`, i.e. `if use_portsData then nPorts else 0`.

HansOlsson, 2018-06-07:
> Edit: That would just lead to e.g. portsData_diameter_internal and portsData_diameter having different types, causing other type issues in the connect equations.

If we assume that connect-equations are removed before checking the sizes/types it works. However, the adding then removing is done for both modifiers and connect-statements; and thus one could argue that there is no need for the other change either.

perost, 2018-06-07:
>If we assume that connect-equations are removed before checking the sizes/types it works. However, the adding then removing is done for both modifiers and connect-statements; and thus one could argue that there is no need for the other change either.

Looking at the flattening section of the 3.4 spec it actually seems like connect-statements are not in fact added and removed. The last paragraph in 5.6.2 says:
>This leads to a flattened equation system, except for connect and transition statements. These have to be transformed as described in Chapter 9 and Chapter 17. This may lead to further changes in the instance tree [e.g. from expandable connectors (section 9.1.3)] and additional equations in the flattened equation system [e.g. connect equations (section 9.2), generated equations for state machine semantics (section 17.3.4)].

For conditional components it says:
>Conditional components with false condition are removed afterwards and they are not part of the simulation model. [Thus e.g. parameters don't need values in them. However, type-error can be detected.]

So with this in mind it seems like connect-statements can be removed before type checking them, in which case the fix I proposed earlier would be fine. Then the question is just what the specification means when it says conditional components should be valid. If this does not mean that they should be valid in any way they can be, then this needs to be clarified.

beutlich, 2019-11-04:
Removing milestone since it is not clear yet if and how to fix.