modelica/ModelicaStandardLibrary #2596: Variable used in a dimension in Modelica.Electrical.MultiPhase.Functions.symmetricOrientationMatrix

eshmoylova, 2018-06-07:
The function Modelica.Electrical.MultiPhase.Functions.symmetricOrientationMatrix has new protected variables **nBase** and **oBase**, introduced in 3.2.3:
https://github.com/modelica/ModelicaStandardLibrary/blob/eea74432d6006527b41676547ec2326d8bcbb8f2/Modelica/Electrical/MultiPhase.mo#L2693
https://github.com/modelica/ModelicaStandardLibrary/blob/eea74432d6006527b41676547ec2326d8bcbb8f2/Modelica/Electrical/MultiPhase.mo#L2696

**nBase** is used as a dimension of **oBase**, but it is a variable. According to the Specification 3.4, Section 12.2:
>The dimension sizes not declared with (:) of each array result or array local variable [i.e., a non-input components] of a function must be either given by the input formal parameters, or given by constant or parameter expressions, or by expressions containing combinations of those (Section 12.4.4).

**nBase** must be declared as a parameter.

It is a regression comparing to 3.2.2.

beutlich, 2018-06-08:
See a7f7cffe2dec9d3bacb0a9c8da8afa19819d4fc7 for #1997.

HansOlsson, 2018-06-08:
As far as I recall "input formal parameters" just means "inputs" to functions; see 12.1.1 - i.e. it is a "parameter" when calling the function - but not a "parameter" in Modelica. I believe that is an influence from computer science and other languages, and I agree that it is problematic.

So, `m `is an input and therefore "input formal parameter", but `nBase `isn't; using 
`Modelica.SIunits.Angle oBase[numberOfSymmetricBaseSystems(m)]=o[1:mBase:m];`

eshmoylova, 2018-06-11:
I second @HansOlsson's suggestion. It would solve the issue.

beutlich, 2018-06-11:
What about `Modelica.SIunits.Angle oBase[:]=o[1:mBase:m];`?

HansOlsson, 2018-06-11:
> What about Modelica.SIunits.Angle oBase[:]=o[1:mBase:m];?

I think that is less good, since the size will depend on mBase - which has similar issue as nBase.

PR #2612 (2018-06-12): Fix variability of oBase dimension
Change from protected variable to input formal parameter

close #2596
commit 3cfef334 (2018-06-12): close #2596: Fix variability of oBase dimension (#2612)

Change from protected variable to input formal parameter
closed 2018-06-12 by beutlich