modelica/ModelicaStandardLibrary #2603: New library for generation of images in documentation

tobolar, 2018-06-11:
I have re-generated a couple of images (pictures) for version v3.2.3, especially regarding Modelica.Mechanics.MultiBody. For that reason, it was necessary to define new models in some cases in order to reconstruct the situation of the original images.
I think this can happen in the future again and again, and maybe also for another libraries. So it would be reasonable to my opinion to have a particular library collecting models for such particular reason, thus simplificating the images' regeneration.

So my question is
1. does it make sense to have such a library,
2. if yes, where to define it. 
   1. standalone or
   2. as a part of another library, e.g. ModelicaTest. The drawback here is that the models for images' generation do not necessarilly need to be simulatable.

dietmarw, 2018-06-11:
What about putting this single-file library under `Resource/Images/ImageCreation.mo` or similar?

beutlich, 2018-06-11:
It is not only for the diagram view but also for the 3D view (see Resources/Documentation/Mechanics/Figure_PlanarLoopAnalytic.ppt) or plotted figures.

HansOlsson, 2018-06-11:
I think I would prefer to have it as a new top-level library; so that people only using MSL will not see this library (but developers clearly see it so we everything is put in the same place).

dietmarw, 2018-06-11:
So top-level `ModelicaDocumentation.mo` perhaps?

sjoelund, 2018-06-11:
I would prefer to have it within some other library; otherwise there are so many small top-level names being used if one looks at the repository.

christiankral, 2019-07-11:
> I think I would prefer to have it as a new top-level library; so that people only using MSL will not see this library (but developers clearly see it so we everything is put in the same place).

On one hand I understand this argument very well, as I think that the user does not need to stumble into directories and files if there is no need for it. On the other hand it does make sense to have all the resources in one place in order to avoid copy + paste and file sync work which is always fault prone and -- finally unnecessary work. 

The actual structure of the ModelicaStandardLibrary repository is:

```
├── Complex.mo
├── CONTRIBUTING.md
├── LICENSE
├── Modelica
│   ├── Resources
│   ├── ├── Images
│   ├── ...
├── ModelicaReference
├── ModelicaServices
├── ModelicaTest
├── ModelicaTestConversion4.mo
├── ModelicaTestOverdetermined.mo
├── ObsoleteModelica4.mo
└── README.md
```

I think we should also consider that some figures are / will be not created through a Modelica model, but for example by scripts created by:

* Python
* Julia
* Octave
* Matlab
etc.

So the different scripts / sources shall mapped be to the structure of the MSL. I guess it does not make sense to build redundant structures for each script language. It may thus make sense to have one directory structure hosting all the different scripts, Modelica and PNG files.

1. Option: Add Images library as directory structure to `Resources` (actually `Modelica/Resources`):

```
Resources/
├── BuildProjects
├── C-Sources
├── Data
├── Documentation
├── Images
├── ├── package.mo
├── ├── package.order
├── ├── Blocks
├── ├── ├── package.mo
├── ├── ├── package.order
├── ├── ├── Sources
├── ├── ├── ├── package.mo
├── ├── ├── ├── package.order
├── ├── ├── ├── LogFrequencySweep_1.mo
├── ├── ├── ├── LogFrequencySweep_2.jl
├── ├── ├── ├── LogFrequencySweep_3.m
├── ├── ├── ├── LogFrequencySweep_1.png
├── ├── ├── ├── LogFrequencySweep_2.png
├── ├── ├── ├── LogFrequencySweep_3.png
├── ├── ├── ...
├── Library
├── ReleaseNotes
└── Scripts
```

2. Option: Alternatively, `LogFrequencySweep` could be a directory (package) with is containing all the relevant files (Modelica / scripts / PNG): This allows a much clearer structure, however.

```
├── ├── ├── Sources
├── ├── ├── ├── package.mo
├── ├── ├── ├── package.order
├── ├── ├── ├── LogFrequencySweep
├── ├── ├── ├── ├── package.mo
├── ├── ├── ├── ├── package.order
├── ├── ├── ├── ├──  LogFrequencySweep_1.mo
├── ├── ├── ├── ├──  LogFrequencySweep_2.jl
├── ├── ├── ├── ├──  LogFrequencySweep_3.m
├── ├── ├── ├── ├──  LogFrequencySweep_1.png
├── ├── ├── ├── ├──  LogFrequencySweep_2.png
├── ├── ├── ├── ├──  LogFrequencySweep_3.png
```

3. Option: Modelica package in one file under `Images`, image and script files in directory structure
  * Keep all Modelica models in one file `Images.mo` (more difficult to track changes, not my favorite) 
  * Use the existing directory of `Images` structure to host PNG and additional script files (Python, Julia, Octave, Matlab, ...)

4. Option: Move the `Images` package directory two levels up, hosting it on the root level. In this case all Modelica and script files could be hosted in (the directory structure of) `Images`, but the PNG files are hosted at the same place as before. The package name `Image` may be subject to discussion.

Further options may be discussed...