modelica/ModelicaStandardLibrary #2604: Modelica.Mechanics.Rotational.Examples.GenerationOfFMUs

beutlich, 2018-06-11:
Whereas MSL v3.2.2 has nicely drawn connection lines to the springDamper component of Modelica.Mechanics.Rotational.Examples.GenerationOfFMUs

[image]

it is no longer the case in v3.2.3-beta.2

[image]

Is this really the intended behaviour?

tobolar, 2018-06-11:
Well this comes due to the adaption of icon and of connectors position of AngleToTorqueAdaptor on new GeneralAngleToTorqueAdaptor. Since both adaptors should be compatible to each other I think this is a drawback of it.
Not nice but necessary.

beutlich, 2018-06-11:
Well, we still can adapt Modelica.Mechanics.Rotational.Examples.Utilities.SpringDamper to match the connector alignment.

But I prefer to have a wider connector distance in the new adaptors since it looks rather sqeezed (while still having vertical space available).

tobolar, 2018-06-11:
I have no preference.

beutlich, 2018-06-11:
OK then, I propose to use the same connector alignment as the MultiplexN/DeMultiplexN blocks of Modelica.Blocks.Routing for the old and new adaptors.

tobolar, 2018-06-11:
1. Do you mean the particular model(s) Multiplex/DeMultiplex or rather MultiplexN/DeMultiplexN with N being 1..6?.
2. And do you really mean to apply it for both the old and new adaptors?

beutlich, 2018-06-11:
I will work on a PR and let you review it.

beutlich, 2018-06-12:
After spending too much spare time on old/new adaptors I tend to leave it as is. OK for you?

tobolar, 2018-06-13:
Fine for me as well.

closed 2018-06-13 by tobolar