modelica/ModelicaStandardLibrary #2626: va_copy must call va_start on its source argument

DagBruck, 2018-06-13:
My reading of the C specification is that you must call va_start on the second argument of va_copy, before you can use it in va_copy. Then of course you must also call va_end on that argument. This also happens around line 5225.

https://github.com/modelica/ModelicaStandardLibrary/blob/060e29284c844ebd9dd52f341213b9ce5ea4dc7e/Modelica/Resources/C-Sources/ModelicaMatIO.c#L1468

Btw, it is unclear if we use VA_COPY (a macro) or va_copy (as on line 1468).

DagBruck, 2018-06-13:
Note: things like this are found by CppCheck, http://cppcheck.net.

sjoelund, 2018-06-13:
> Btw, it is unclear if we use VA_COPY (a macro) or va_copy (as on line 1468).

`va_copy` is also a macro (or builtin; or both)...

beutlich, 2018-06-13:
The one on line 1468 is an upstream issue for http://matio.sf.net/.
The other one around line 5225 is an upstream issue for https://github.com/weiss/c99-snprintf.

beutlich, 2018-06-14:
* line 1468: strdup_vprintf is only called from strdup_printf, which calls va_start just before strdup_vprintf.
* line 5222: rpl_vasprintf (or rather mat_vasprintf) is not called at all

I'd like to close this as non-issue then.

closed 2018-06-14 by beutlich

DagBruck, 2018-06-15:
Well, we have customers that complain about such warnings in the C code. But it helps if we can blame MA or possibly some third party.

weiss, 2018-06-21:
> The other on line 5222 is an upstream issue for https://github.com/weiss/c99-snprintf.

I [don't think][1] that's true.

[1]: https://github.com/weiss/c99-snprintf/issues/2#issuecomment-398369437