modelica/ModelicaStandardLibrary #2628: Text in icons of sensors

tobolar, 2018-06-13:
It should be harmonised which text is used in sensors.
- quantity name, like _V_ in `Thermal.FluidHeatFlow.Sensors.VolumeFlowSensor`
- quantity unit, like _A_ in `Electrical.Analog.Sensors.CurrentSensor`
- quantity text, like _power_ in `Mechanics.Rotational.Sensors.PowerSensor`
- something else
- a combination of abovementioned

Refs #2627

christiankral, 2018-06-21:
In electrical engineering we use the unit within the circular sensor symbol to indicate which meter it is. Therefore, for a power meter we use "W" to indicate it as watt-meter. How do other engineers indicate their sensors? I do not know.

christiankral, 2018-06-21:
This issue is actually also linked to https://github.com/modelica/ModelicaStandardLibrary/pull/2471#issuecomment-368490019, where I proposed color code (128,128,128) for descriptive text of an icon. The text of the unit / quantity / whatever should actually also use the proposed color code.

christiankral, 2018-11-19:
Even the electrical sensors icons are not designed in the same way: 

[image]

Thermal sensor of `Modelica.Thermal.FluidHeatFlow` and `Modelica.Thermal.HeatTransfer` shall also be harmonized:

[image]

dietmarw, 2018-11-19:
This probably could be solved elegantly by having a proper base sensor icon class which allows for the label (*e.g.,* P, I, V, dp, H, M) and maybe the optional `m=...`  parameter.

dietmarw, 2018-11-19:
As for unit vs. quantity. Maybe the display of `[A]` would make it more obvious that it is a quantity. Also since the senors give a signal out it might help to actually have the unit and not quantity displayed. Since that is what the signal is "attached" to.

HansOlsson, 2018-11-19:
> Quantity might be less ambiguous than unit in case of unit ≠ displayUnit.

I agree that case of unit ≠ displayUnit is important, but I am unsure about the implication - consider Modelica.Fluid.Examples.DrumBoiler.DrumBoiler

christiankral, 2019-06-16:
> It should be harmonised which text is used in sensors.
>
>     * quantity name, like _V_ in `Thermal.FluidHeatFlow.Sensors.VolumeFlowSensor`
>
>     * quantity unit, like _A_ in [...]

I totally agrree. We need to come up with a decision. The majority of sensors in the MSL places a black label of the quantity next to the output(s) of the sensor. Personally I am used to indicate an Ampere meter by "A" not "i". For the sake of consistency I have no issue to switch to "i".

@AHaumer Do you have any objections against "i" for an Ampere meter?

christiankral, 2019-06-16:
Onve we come up with a decision we need to write it down in `Modelica.UsersGuide.Conventions.Icons`

AHaumer, 2019-06-19:
I dislike "i" for Amperemeter - lots of electrical engineers won't understand. They're used to read "A".

dietmarw, 2019-06-19:
So we should probably unify it to have the **unit** displayed in the icon? T

christiankral, 2019-06-20:
So the SI unit is also OK to me. 

If we speak of unifying it, is it meant to unify it in all sensors of the MSL? Or do we mean only the electrical sensors?  

There are yet some other open questions we shall agree on when unifying the sensors. 

A. What color shall the unit have? We have nothing on this question in the icon guidelines yet. 

  1. Black color
  2. Gray color (e.g. {192,192,192} or {128,128,128})
  3. Domain color (i.e. blue {0,0,255} in case of analog sensors)

B. In case ft is a single output sensor: shall the SI units be placed inside the circle or next to output? 

C. In case of a multi output sensor: shall the SI unit be placed next to every signal output; there may, however, not be sufficient space... 

E. Shall non output connectors also have a label, as `resolve` in the example below? I guess no, since the tooltip of the tool usually resolves this issue. 

[image]

Here some color / location / size examples:

[image]

maltelenz, 2019-06-24:
With regards to the coloring (point A), I'd prefer if we used a dark gray, for example {64, 64, 64}. It's easy to see, without being glaring (like black and blue is).

(Just as a side note, in the MSL we ship in Wolfram SystemModeler, we change all label colors, including %name and parameters, into {64, 64, 64}, because we consider the current default coloring to ugly and unprofessional.)

beutlich, 2019-06-29:
Since there haven't been any objections (neither other proposals), I propose to move on with the following suggestions.

 - A: Use textColor={64,64,64} for the SI unit in sensors (as proposed by @maltelenz and also is used for the sensor arrows).
- B: Single-sensors have the SI unit displayed in the circle.
- C: Multi-sensors have the Si units displayed next to the output connector.
- E: Not relevant for this issue.

@christiankral Can you move on and prepare a PR for it?

christiankral, 2019-06-30:
I will prepare a PR -- I suppose for all packages of the MSL. This will hopefully not cause any conflicts as there may be some open PRs. 

Additionally, I will create a PR adding the applied rules to the MSL User's Guide.

beutlich, 2019-06-30:
> I will prepare a PR

Thanks.

> This will hopefully not cause any conflicts as there may be some open PRs.

Conflicts mainly occur in ModelicaTestConversion4.mo or ConvertModelica_from_3.2.3_to_4.0.0.mos which are easy to handle in case of additions.

tobolar, 2019-09-03:
@christiankral Shall I proceed with `Modelica.Mechanics.MultiBody` or are you working on it? Consequently, the descriptive text of a single output shall be removed, e.g. "force" in `Modelica.Mechanics.MultiBody.Sensors.CutForce`. The text by multiple outputs shall be changed to units.

christiankral, 2019-09-04:
@tobolar Please go ahead with `Modelica.Mechanics.MultiBody`. I will have little time resources to work on this anytime soon.

tobolar, 2019-09-04:
@christiankral
I'm not satisfied with different text size used within the Rotational and Translational sensors, see image below.
Also, the connection lines shall be interrupted where there is a text, as indicated in `Modelica.UsersGuide.Conventions.Icons`:
> If there is a connector located at the top icon boundary and it is obvious that this connector influences the model behavior compared to a similar model without such connector, then a line from the connector to the actual icon shall be avoided to keep the design straight, see Fig. 4. **If it is required to use a line indicating the connector dependency, then the line shall be interrupted such that this line does not interfere with component name.** 

I suppose we could apply this rule to another text in icons as well.

So I'm going to adapt the icons accordingly provided you have no objections.

[image]

tobolar, 2019-09-04:
This is how it will looks like (Dymola 2020):
[image]

christiankral, 2019-09-05:
@tobolar OK, I am fine with all your proposals.

tobolar, 2019-09-05:
@christiankral Please add your review on #3107. The changes there comply with the picture above.

dietmarw, 2020-01-24:
Can this be closed?

closed 2020-01-24 by beutlich