modelica/ModelicaStandardLibrary #2636: Parameters' description string missing or not starting with upper case in Electrical

tobolar, 2018-06-14:
The following components from `Modelica.Electrical` are affected.
- [ ] Modelica.Electrical.Analog.Examples.CauerLowPassAnalog
    - l1: Not start with upper case: filter coefficient I1
    - l2: Not start with upper case: filter coefficient I2
    - c1: Not start with upper case: filter coefficient c1
    - c2: Not start with upper case: filter coefficient c2
    - c3: Not start with upper case: filter coefficient c3
    - c4: Not start with upper case: filter coefficient c4
    - c5: Not start with upper case: filter coefficient c5
- [ ] Modelica.Electrical.Analog.Examples.CauerLowPassOPV
    - l1: Not start with upper case: filter coefficient i1
    - l2: Not start with upper case: filter coefficient i2
    - c1: Not start with upper case: filter coefficient c1
    - c2: Not start with upper case: filter coefficient c2
    - c3: Not start with upper case: filter coefficient c3
    - c4: Not start with upper case: filter coefficient c4
    - c5: Not start with upper case: filter coefficient c5
- [ ] Modelica.Electrical.Analog.Examples.CauerLowPassSC
    - l1: Not start with upper case: filter coefficient i1
    - l2: Not start with upper case: filter coefficient i2
    - c1: Not start with upper case: filter coefficient c1
    - c2: Not start with upper case: filter coefficient c2
    - c3: Not start with upper case: filter coefficient c3
    - c4: Not start with upper case: filter coefficient c4
    - c5: Not start with upper case: filter coefficient c5
- [ ] Modelica.Electrical.Analog.Examples.HeatingNPN_OrGate
    - CapVal: Description string missing
    - tauVal: Description string missing
- [ ] Modelica.Electrical.Analog.Examples.HeatingPNP_NORGate
    - CapVal: Description string missing
    - tauVal: Description string missing
- [ ] Modelica.Electrical.Analog.Examples.Rectifier
    - f: Not start with upper case: line frequency
    - LAC: Not start with upper case: line inductor
    - Ron: Not start with upper case: diode forward resistance
    - Goff: Not start with upper case: diode backward conductance
    - Vknee: Not start with upper case: diode threshold voltage
    - IDC: Not start with upper case: load current
- [ ] Modelica.Electrical.Analog.Examples.ResonanceCircuits
    - res: Not start with upper case: f/fResonanace
- [ ] Modelica.Electrical.Analog.Examples.OpAmps.InvertingSchmittTrigger
    - vHys: Not start with upper case: (positive) hysteresis voltage
- [ ] Modelica.Electrical.Analog.Examples.OpAmps.SchmittTrigger
    - vHys: Not start with upper case: (positive) hysteresis voltage
- [ ] Modelica.Electrical.Analog.Basic.OpAmpDetailed
    - Ts: Not start with upper case: sampling time
    - Avd0_val: Not start with upper case: differential mode gain
    - Avcm_val: Not start with upper case: common mode gain
- [ ] Modelica.Electrical.Analog.Basic.VariableCapacitor
    - Cmin: Not start with upper case: lower bound for variable capacitance
    - UIC: Description string missing
- [ ] Modelica.Electrical.Analog.Basic.VariableInductor
    - Lmin: Not start with upper case: lower bound for variable inductance
    - UIC: Description string missing
- [ ] Modelica.Electrical.Analog.Basic.Potentiometer
    - useRinput: Not start with upper case: use input for 0<r<1 (else constant)
- [ ] Modelica.Electrical.Analog.Interfaces.ConditionalHeatPort
    - useHeatPort: Not start with upper case: =true, if heatPort is enabled
- [ ] Modelica.Electrical.Analog.Lines.OLine
    - useHeatPort: Not start with upper case: =true, if HeatPort is enabled
    - T_ref: Description string missing
- [ ] Modelica.Electrical.Analog.Lines.M_OLine
    - useHeatPort: Not start with upper case: =true, if HeatPort is enabled
    - T_ref: Description string missing
- [ ] Modelica.Electrical.Analog.Lines.ULine
    - useHeatPort: Not start with upper case: =true, if HeatPort is enabled
    - T_ref: Description string missing
- [ ] Modelica.Electrical.Analog.Semiconductors.NPN
    - EMin: Not start with upper case: if x < EMin, the exp(x) function is linearized
    - EMax: Not start with upper case: if x > EMax, the exp(x) function is linearized
[... 195 lines cut ...]
- [ ] Modelica.Electrical.PowerConverters.Interfaces.Enable.Enable
    - useConstantEnable: Not start with upper case: true = disabled boolean input, use constantEnable
- [ ] Modelica.Electrical.PowerConverters.Interfaces.Enable.Enable1m
    - useConstantEnable: Not start with upper case: true = disabled boolean input, use constantEnable
- [ ] Modelica.Electrical.QuasiStationary.SinglePhase.Sources.FrequencySweepVoltageSource
    - phi: Not start with upper case: phase shift of the source
- [ ] Modelica.Electrical.QuasiStationary.MultiPhase.Basic.PlugToPins_p
    - m: Not start with upper case: number of phases
- [ ] Modelica.Electrical.QuasiStationary.MultiPhase.Basic.PlugToPins_n
    - m: Not start with upper case: number of phases
- [ ] Modelica.Electrical.QuasiStationary.MultiPhase.Sensors.PowerSensor
    - m: Not start with upper case: number of phases
- [ ] Modelica.Electrical.QuasiStationary.MultiPhase.Interfaces.AbsoluteSensor
    - m: Not start with upper case: number of phases
- [ ] Modelica.Electrical.Spice3.Examples.Spice3BenchmarkDifferentialPair
    - MOD1: Description string missing
- [ ] Modelica.Electrical.Spice3.Examples.Spice3BenchmarkMosfetCharacterization
    - MOD1: Description string missing
- [ ] Modelica.Electrical.Spice3.Examples.Spice3BenchmarkRtlInverter
    - Q11: Description string missing
- [ ] Modelica.Electrical.Spice3.Examples.Spice3BenchmarkFourBitBinaryAdder
    - DMOD: Description string missing
    - QMOD: Description string missing
- [ ] Modelica.Electrical.Spice3.Internal.MOS2
    - m1: Description string missing

beutlich, 2018-06-14:
Do you have a script that generates this list?

christiankral, 2018-07-01:
@ThomasBoedrich I updated the documentation string of `sigma` in `Modelica.Magnetic.FluxTubes.Material.HysteresisTableData.BaseData` to:

`parameter SI.Conductivity sigma = 1 "Electric conductivity";`

I suppose this is OK as simply took the explanation from the HTML documentation.

christiankral, 2018-07-01:
@tobolar 
The protected parameter arrays 

```modelica
  parameter Real Ptable[:]={1E-6,1845,3549,5325,7521,9372,11010,12930,
      14950,16360,18500,18560,20180,22170};
  parameter Real Itable[:]={11.0,11.20,12.27,13.87,16.41,18.78,21.07,
      23.92,27.05,29.40,32.85,32.95,35.92,39.35};
  parameter Real ntable[:]={1500,1496,1493,1490,1486,1482,1479,1475,1471,
      1467,1462,1462,1458,1453};
  parameter Real ctable[:]={0.085,0.327,0.506,0.636,0.741,0.797,0.831,
      0.857,0.875,0.887,0.896,0.896,0.902,0.906};
  parameter Real etable[:]={0,0.7250,0.8268,0.8698,0.8929,0.9028,0.9064,
      0.9088,0.9089,0.9070,0.9044,0.9043,0.9008,0.8972};
```
in `Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.AIMC_withLosses` and `Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_withLosses` are NOT included in your list; however, I do not know if this a consequence of the array type or the protected section...

tobolar, 2018-07-02:
@christiankral Protected parameters are excluded in the list. But you are of course free to add the parameter decription to them.

tobolar, 2018-07-02:
@kristinmajetta, @christiankral Would you also mark proceeded classes in the check list of this issue?
If you mean it's too much effort (well there are really a lot of classes concerned), I would change the check list to simple bullet list.

kristinmajetta, 2018-07-03:
marked the proceeded classes from Modelica.Electrical.Analog, Modelica.Electrical.Digital and Modelica.Electrical.Spice3

beutlich, 2019-02-18:
@casella Can you please care for the Modelica.Fluid descriptions as stated above.

christiankral, 2019-06-16:
Done from my side.

beutlich, 2019-06-16:
> 
> 
> Done from my side.

Only Fluid and Media left to do, right?

christiankral, 2019-06-16:
I missed `Modelica.Magnetic.FluxTubes` previously; now covered by #2980.

Otherwise, Fluid, Media and Math are missing.

beutlich, 2019-10-22:
@casella Reminder: Can you please update the description comments in Fluid as reported?

beutlich, 2019-12-21:
> 
> 
> @casella Reminder: Can you please update the description comments in Fluid as reported?

I did so in #3281. Please review.

PR #3281 (2020-01-07): Add/fix description comments in Fluid
Fix the remaining parameter descriptions to finally resolve #2636.
closed 2020-01-07 by beutlich