modelica/ModelicaStandardLibrary #2646: Change in World-object causes regressions

HansOlsson, 2018-06-18:
The correction for #2432 causes regressions in a commercial library (FlightDynamics); and in theory there could be other similar cases.

Basically if a library made their own world-object (by copying World from MSL and adding parts) and used it as "inner MyWorld world;" the "outer World world;" in MSL no longer matches the inner component due to the new parameter for animateGround.

I don't see that we have stated that making your own inner-objects is not fully supported.

One possibility would be to accept it and explain that it is not supported.

tobolar, 2018-06-18:
Focusing on 
> by copying World from MSL

I suppose you don't mean *extending*.
> One possibility would be to accept it and explain that it is not supported.

seems then to be the simplest and reliable solution.

beutlich, 2018-06-18:
Adding a new parameters always can lead to regressions, since existing user classes that extend from the this class might break. We have it stated in [Modelica.UsersGuide.ReleaseNotes.VersionManagement](https://doc.modelica.org/help/Modelica_UsersGuide_ReleaseNotes.html#Modelica.UsersGuide.ReleaseNotes.VersionManagement):
> Introducing a new name in the public section of a class (model, package, ...) or in any section of a partial class is **not** allowed.

However, we haven't paid attention to the rule for years and somehow considered additions to public sections as legal, even in maintenance releases.

Therefore, I am in favour to accept the regression and mention it in the release notes.

HansOlsson, 2018-06-18:
> Adding a new parameters always can lead to regressions, since existing user classes that extend from the this class might break.

True, but this inner/outer mismatch is another failure mode (compared to partial models) that we had not considered earlier. I hope we are not adding too many new parameters to partial classes. (Clearly it is possible to extend a non-partial class as well.)

beutlich, 2018-06-18:
Have a look at Modelica.Thermal.FluidHeatFlow.Interfaces.Partials.SinglePortLeft and 7f29c1ef3919dfcbe85558df5296585173993dfb (around line 2971+)? Acceptable or not?

tobolar, 2018-06-18:
Are we talking about extending the World or copying it? I think copying is always on users risk and he/she has to care about upgrade on his/her own.

HansOlsson, 2018-06-18:
> Are we talking about extending the World or copying it? I think copying is always on users risk and he/she has to care about upgrade on his/her own.
 
Copying World, and as far as I know we haven't stated that you aren't supposed to do this.

HansOlsson, 2018-06-18:
For the longer term we could consider if we should split the "World" into two models: one that is used as "outer World"; and one (or several) models extending from that one used for the inner one - it can include optional effects like coordinate system and ground visualization that isn't needed for the outer world.

HansOlsson, 2018-06-25:
> Introducing a new name in the public section of a class (model, package, ...) or in any section of a partial class is not allowed.

Note that this is only stated under the "maintenance branch", e.g. 3.2.2 versionbuild 4 - and thus does not apply to this release. (I realized this when commenting on #2662 )

beutlich, 2018-06-25:
There seems to be agreement to leave the MultiBody.World as is and document the change. I will document it in the release notes (and leave this issue open as a reminder).

PR #2566 (2018-08-05): Finalise the Release Notes 3.2.3
started to fill release notes 3.2.3

close #2565
close #2646 
close #2662
closed 2018-08-05 by beutlich