modelica/ModelicaStandardLibrary #2654: Changes to Electrical.Analog icons in 3.2.3

eshmoylova, 2018-06-19:
The following commit made a lot of changes to icons in Electrical.Analog: https://github.com/modelica/ModelicaStandardLibrary/commit/232813ba512ea89e71a9f08d9b02c278b8c97756

There are a couple of changes that I am not sure whether they were intentional or not, and the description of the commit(s) does not provide many details. @christiankral (or anyone else who is familiar with the changes), can you please verify that the following icons look as intended and not regressions?
Electrical.Analog.Basic.HeatingResistor (The arrow disappeared)
3.2.2:
[image]
3.2.3:
[image]
Electrical.Analog.Basic.M_Transformer (The dotted line disappeared)
3.2.2:
[image]
3.2.3:
[image]
Electrical.Analog.Ideal.ControlledIdealCommutingSwitch had port labels in 3.2.2 but not in 3.2.3.
3.2.2:
[image]
3.2.3:
[image]
Electrical.Analog.Ideal.ControlledIdealIntermediateSwitch had port labels in 3.2.2 but not in 3.2.3.
3.2.2:
[image]
3.2.3:
[image]
Electrical.Analog.Ideal.IdealCommutingSwitch had port labels in 3.2.2 but not in 3.2.3.
3.2.2:
[image]
3.2.3:
[image]
Electrical.Analog.Ideal.IdealIntermediateSwitch used to have more lines and port labels.
3.2.2:
[image]
3.2.3:
[image]

christiankral, 2018-06-20:
The Modelica.Electrical.Analog.Ideal.IdealIntermediateSwitch does not look as intended. For some reason the icon appears on the diagram instead of the icon layer. I will fix this. All other icons look OK.

beutlich, 2018-06-20:
@christiankral Please explicitly confirm that

- [ ] the removed arrow of the HeatingResistor
- [ ] the removed dotted line in the transformer
- [ ] the removed vertical lines to the top pin in the switches

are by purpose!

christiankral, 2018-06-20:
Agree: all these changes are made on purpose:
- The arrow is not required as the model is fully compatible to the regular resistor model which does NOT have an arrow
- The dotted line in the transformer model -> minimalistic approach
- The vertical lines to the switches are removed -> minimalistic approach, since it is clear from the symbol which connectors are triggering the switch

eshmoylova, 2018-06-20:
@christiankral What about port labels on switches? Are they also removed as part of minimalistic approach?

christiankral, 2018-06-20:
@eshmoylova Yes, this follows the same minimalistic approach: The port labels are removed, since tooltips (in Dymola and OpenModelica) show the instance names of the connectors. However, we do not use port labels in other electrical components, so it makes sense to have these obsolete port labels removed.

beutlich, 2018-06-20:
Thanks. All questions clarified.

closed 2018-06-20 by beutlich

tobolar, 2018-06-21:
> The port labels are removed, since tooltips (in Dymola and OpenModelica) show the instance names of the connectors.

See also discussion in PR  #2471, which IMO is still open (the discussion, not the PR).