modelica/ModelicaStandardLibrary #2661: Annotation "obsolete" in send/receive blocks

tobolar, 2018-06-20:
In the following blocks there is an _obsolete_ annotation "Obsolete block - use expandable connectors instead":
- Modelica.Blocks.Interfaces.Adaptors.SendReal
- Modelica.Blocks.Interfaces.Adaptors.SendBoolean
- Modelica.Blocks.Interfaces.Adaptors.SendInteger
- Modelica.Blocks.Interfaces.Adaptors.ReceiveReal
- Modelica.Blocks.Interfaces.Adaptors.ReceiveBoolean
- Modelica.Blocks.Interfaces.Adaptors.ReceiveInteger

In some cases where such blocks are used (misused?), there could be used following blocks instead the obsolete ones as well:
- Modelica.Blocks.Routing.RealPassThrough
- Modelica.Blocks.Routing.IntegerPassThrough
- Modelica.Blocks.Routing.BooleanPassThrough

Suggestion is to extend the _obsolete_ annotation to something like "Obsolete block - use expandable connectors instead or Modelica.Blocks.Routing.XXXPassThrough", with XXX being Real, Boolean or Integer.

beutlich, 2018-06-20:
I used the text from the documentation when editing the obsolete message. Just go ahead, if you think it makes things clearer, but these adaptors are obsolete since c4609eca18ddf4efb261c73eb107a6dae1e8ab0a (for MSL v3.0.1).

beutlich, 2018-08-04:
OK, no more issue then.

closed 2018-08-04 by beutlich