modelica/ModelicaStandardLibrary #2662: Calrify if broken backward compatibility is acceptable

beutlich, 2018-06-20:
Commit ba52d0a40e1973805c0e300da8cd23a212cf0fa3 for #2364 adds a new constant (C_default) to a partial class (PartialMedium), which - strictly seen - breaks the desired backward compatibility of MSL v3.2.3. This is a discussion issue, if we accept it nevertheless and document it in the release notes or, should revert it to not break compatibility.

@MartinOtter @HansOlsson @hubertus65 Please comment and advice.

HansOlsson, 2018-06-25:
To me it is acceptable for 3.2.3 (but would not be ok for a maintenance release), and should document it.

Regardless of whether it is a partial or not a new constant can break other classes, and I don't see that we have made any special rules for partial classes (or partial packages).

However, when documenting this we should consider both of the following failure modes:

- The obvious one is that if a class extends from PartialMedium  and has an element named "C_default" it will no longer work.
- The non-obvious one is that if a user made their own medium without extending PartialMedium it will no longer be sub-type compatible. This was the problematic case in #2646 (but using inner/outer); and since the "outer" parameters were used it wasn't just sub-type check.

To avoid the second case we should also strongly suggest that if a model in MSL has sub-typing w.r.t. to a class A (due to replaceable declaration or inner/outer) users of MSL should extend from A; and not copy A.

beutlich, 2018-06-25:
@HansOlsson Thanks for the comment. I will document in the release notes (and leave this open now as a reminder).

PR #2566 (2018-08-05): Finalise the Release Notes 3.2.3
started to fill release notes 3.2.3

close #2565
close #2646 
close #2662
closed 2018-08-05 by beutlich