modelica/ModelicaStandardLibrary #2672: Abrupt adaptor

Claytex, 2018-06-26:
Hi, this was sent to DS but they asked me to present the issue here hoping it can be resolved please.

Hi, we are unable to get this simple model to run and I was wondering if you could suggest what the user is doing wrong please.

[Test_MSL_Abrupt_adaptorTotal.zip](https://github.com/modelica/ModelicaStandardLibrary/files/2137776/Test_MSL_Abrupt_adaptorTotal.zip)

HansOlsson, 2018-06-27:
To clarify: 
The Adrupt_adaptor states that it handles both flow directions and allowFlowReversal is true; but here it fails to simulate since the pressure is higher at the left.

If you change the ramp to have offset=1.0e5 and startTime 0.5s the simulation will stop at 0.527s when the boundary.p exceeds boundary1.p.

The problem can also be reproduced with Modelica.Fluid.Examples.Explanatory.MomentumBalanceFittings by changing leftBoundary1.p to 1.2e5.
(It seems ModelicaTest.Fluid.TestComponents.Fittings.TestSuddenExpansion is a duplicate of this.)

HansOlsson, 2018-06-28:
As far as I can tell the reason is that Modelica.Fluid.Fittings.BaseClasses.QuadraticTurbulent.pressureLoss_m_flow_totalPressure always return a negative value for cases like this; regardless of mass-flow.

E.g. in Dymola I could plot:
```
plotArray(linspace(-2,2,1000),
Modelica.Fluid.Fittings.BaseClasses.QuadraticTurbulent.pressureLoss_m_flow_totalPressure(linspace(-2,2,1000),1e3,1e3,1e3,1e3,Modelica.Fluid.Fittings.BaseClasses.QuadraticTurbulent.LossFactorData(D_Re=0.1, Re_turbulent=100.0, c0=30.0, diameter_a=0.1, diameter_b=0.2, zeta1=0.5625, zeta1_at_a=true, zeta2=0.4029637244338282,   zeta2_at_a=true, zetaLaminarKnown=true),0.01))
```
And it is always negative.

I don't know if the inputs are wrong, it is not applicable, or the underlying function has some error.

beutlich, 2018-08-04:
@casella Can you please have a look at this.

beutlich, 2019-11-04:
Removing milkestone after missing feedback from library officers.