modelica/ModelicaStandardLibrary #2673: Inconsistent initial conditions in fullRobot

eshmoylova, 2018-06-26:
The example model Modelica.Mechanics.MultiBody.Examples.Systems.RobotR3.fullRobot was already mentioned in #1741 as problematic for initialization. But it was considered only as a problem for regression testing, and, as far as I can see, no changes were made. The model is still used as an example. The inconsistency in initialization appears to be significant but is ignored in Dymola and OpenModelica, and the model is still listed as an example.

The initial conditions generated by Dymola violate one of the equations. The equation in question comes from Modelica.Mechanics.Rotational.Components.SpringDamper:
```modelica
      tau_c = c*(phi_rel - phi_rel0);
```
In the model (for one of the axis) the equation (as reported in the generated listing of flat Modelica code in .mof file) becomes:
```modelica
    axis2.gear.spring.tau_c = axis2.gear.spring.c*(axis2.gear.spring.phi_rel-
      axis2.gear.spring.phi_rel0);
```
In translated Modelica code (dsmodel.mof) with the parameters substituted the equation becomes:
```modelica
  axis2.gear.spring.tau_c := 8.0*axis2.gear.spring.phi_rel;
```
The computed initial values are:
```
  axis2.gear.spring.tau_c = 2.19321
  axis2.gear.spring.phi_rel = 15.7077
```
These values obviously violate the equation above but no error is reported.

In OpenModelica, the same equation binding axis2.gear.spring.tau_c and axis2.gear.spring.phi_rel is listed in the translated model. If I select Simulation -> Simulation Setup -> Simulation Flags -> LOG_INIT and LOG_RES_INIT, the values reported in the log are
```
Real axis2.gear.spring.phi_rel(start=0, nominal=0.0001) = 0.274151 (pre: 0)
Real axis2.gear.spring.tau_c(start=0, nominal=1) = 2.19321 (pre: 0)
```
These values satisfy the equation. However, when plotted the value for axis2.gear.spring.phi_rel at 0 is 15.7077. If I export the plot results to a CSV file using Export Variables, the values reported at 0 in the CSV file are the same as in the plots and violate the equations.
[image]

But if I look at the output CSV file generated during the simulation (Simulation -> Simulation Setup -> Output -> Output Format -> csv), the values are the ones that satisfy the equation:
[image]

MapleSim finds the same values for tau_c and phi_rel as Dymola but detects that they cannot satisfy that equation and fails.

It looks like the model has an inconsistency that appears in different tools, which deal in different ways with it. Which is the correct way? Should the model be updated to eliminate the problem?

tobolar, 2018-06-27:
Apparently an issue with units: 0.274151 = 15.7077 * PI / 180
I will check it.

HansOlsson, 2018-06-27:
I don't see that there is an error on the tool-side either for Dymola.

Dymola shows phi_rel as 15.7077 deg in plots - which as noted above is 0.27 rad; and thus consistent with the model.

I don't know if OpenModelica presents initialization results and simulation results in different units.

If there were actual differences:
One possibility for Dymola would be that we first get some values during initialization and then other values at the start of the simulation (ignoring the initial equations). Since Dymola does not normally store the initialization values a difference between initialization values and start of the simulation  values would be a possibility, but I cannot that it happens in this case.  It is possible to have if models using "initial" in the simulation equations can have that behavior - and I consider that bad style.

eshmoylova, 2018-06-27:
Thank you, @tobolar  and @HansOlsson for clarifying.

closed 2018-06-27 by beutlich