modelica/ModelicaStandardLibrary #2675: Temperature jumping to 0K with V_flow = 0

markusandres3DS, 2018-06-28:
Using Dymola 2019 FD01 Dev1 with the model [CoolingCircuit.zip](https://github.com/modelica/ModelicaStandardLibrary/files/2144475/CoolingCircuit.zip) the temperature exhibits discontinuities when the volume flow is set to 0 (temperature jumps to 0K). 

Notes:
- I used MSL Version: 3.2.3, 2018-08-01, build 1 (2018-08-01 12:00:00Z) - that comes with Dymola 2019 FD01 Dev1
- The problem was not existant with MSL 3.2.2 in Dymola 2019 FD01 Dev1 or Dymola 2019.
- The problem is resolved when the relative pressure sensor is removed from the simulation.

beutlich, 2018-06-28:
Which v3.2.3 is it exactly - beta.1, beta.2 or beta.3?

markusandres3DS, 2018-06-28:
Should be beta.2 from the information I got with a mail. Can I verify that in the MSL itself? The package annotation states `revisionId="eea74432d` 2018-06-06 08:20:09 +0200",`

markusandres3DS, 2018-06-28:
Same problem exists in Dymola 2019 with MSL 3.2.3 version from above.

beutlich, 2018-06-28:
Yes, eea74432d is [beta.2](https://github.com/modelica/ModelicaStandardLibrary/releases/tag/v3.2.3-beta.2).

beutlich, 2018-06-28:
Caused by ff158961f91c61ba081dbfb6325badd139151f1a (for #2571) where RelativeSensor was updated to extend from TwoPort.

PR #2682 (2018-08-05): resolves #2675
Switched back from inheritance from TwoPort for sensors.
In the same run anhanced (I hope all) instations to full qualified names.

Close #2675
commit b933e3b6 (2018-08-05): resolves #2675 (#2682)
closed 2018-08-05 by beutlich