modelica/ModelicaStandardLibrary #2679: Duplicity in Blocks.MathBoolean and Blocks.Logical

tobolar, 2018-06-29:
The two blocks `Modelica.Blocks.MathBoolean.RisingEdge` and `Modelica.Blocks.Logical.Edge` seem to have the same functionality.
I understand the application area of the blocks could be different, but shouldn't it be better to e.g. extend `Logical.Edge` from `MathBoolean.RisingEdge`?

Silimar holds for other boolean blocks as well:
- Modelica.Blocks.Logical.And
- Modelica.Blocks.Logical.Or
- Modelica.Blocks.Logical.Xor
- Modelica.Blocks.Logical.Nand
- Modelica.Blocks.Logical.Nor
- Modelica.Blocks.Logical.Not
- Modelica.Blocks.Logical.Change
- Modelica.Blocks.Logical.FallingEdge

beutlich, 2018-06-29:
Did you have a look at #2164? `MathBoolean` was added to supersede `Logical` but still is incomplete.

tobolar, 2018-07-11:
Ok, so it's reasonable to reduce any effort on Logical. Thanks.

closed 2018-07-11 by tobolar