modelica/ModelicaStandardLibrary #2686: Incorrect flow direction in Modelica.Fluid.Examples.AST_BatchPlant.BatchPlant_StandardWater?

HansOlsson, 2018-07-09:
In this model I noticed that B2 has a few drops of water flowing out of the port at the top of the tank, which is plain weird!  I don't think it is any error in symbolic processing in Dymola.

Specifically B2.top_ports[1].m_flow (same as V6.port_a.m_flow) has a negative dip at 741.5878891772804s (followed by a larger positive flow. note: there is a also  a positive "dip" at 3262.7830640209s). 
Physically that should not happen, and top_ports.m_flow has a min-value of 0. 

You have to zoom in to notice that B2.m also decreases slightly at that time from 4.972674415057601 to 4.972673539863315 (so about 13 drops of water).

As far as I can tell the model does not directly try to prevent this flow-situation.

Results using Dymola 2019 (and turning off equidistant time grid and storing in double precision).
(Still present in current beta, and could be reproduced with other solvers as well.)

beutlich, 2019-10-15:
@casella Any feedback?

beutlich, 2020-02-09:
Still present with MSL 4.0.0 Alpha.1 and Dymola 2020.

[image]

SimulationX 4.0 does not show the negativ flow, but also the positive flow region around t=740s.

[image]

SimulationX simulation hangs at about 1913.969, thus I have no comparison plot fo the other time reported for 3262.7830640209s.

casella, 2020-02-12:
I had a look at the issue. At time =781.588, a certain number of valves is opened instantaneously: V5, V6, V19, V20, V24, V25, and pump P2 is also turned on. Ideall, water should instantaneously flow from the pump through the valves. However, the pump P2 contains a volume of 100 ml of water, and the pipe has a volume of about 240 ml. Now, the water model used here is StandardWater, that takes into account the (very small) compressibility of cold water. This very small compressibility triggers an extremely fast transient that reduces the pressure for a few microseconds (I can't actually zoom in enough in Dymola to see this transient), causing the valve to get a reverse delta-p and then sucking some drops of water from the top. 

There are multiple issues here. 

The first is that we are talking about fictitious physical behaviour, becaus in practice opening the valves and turning on the pump will need a lot more than a few microseconds, so this fast dynamics will never be triggered in real life. This would require to use ramps instead of boolean signals, hence different pump and valve models. 

The second problem is that in real life you would first turn on the valve and shortly (but not instantaneously) after open the valves, to make sure the pressure build-up is enough to get the right flow direction. This delay is not modelled in the controller, which is highly idealized.

The third problem is that whenever you connect a valve model to the top port of a tank, you have to ensure that either the valve is closed, or the delta-p is positive. A negative delta-p doesn't mean zero flow rate, it actually means water entering the circuit, which is something that a) Modelica.Fluid cannot represent and b) should never ever happen in practice.

The fourth problem is that this model uses incredibly accurate physical model (including fluid compressibility), coupled to highly idealized controller and actuation models (no delays, no actuator response time). Hence, some kind of artifacts is inevitable.

The only quick fix I can think of now is to tweak the controller so that some time elapses between the moment the pump is turned on, and the moment the valves are opened. I'll see what I can do.

beutlich, 2020-03-02:
Removing milestone.

PR #3408 (2020-03-04): Improved controller of AST_BatchPlant example case
This PR improves the controller of AST_BatchPlant by adding a suitable delay between the activation of the pumps and valves upstream V3 and V6 and their actual opening. This is more realistic, as it allows the pressure to stabilize, and it avoids a very short, but quite large, unrealistic backflow through those valves.

A bit of backflow is still present, but that cannot be eliminated unless the valve opening is a ramp and is not governed by a boolean signal.

This PR fixes #2686
closed 2020-03-04 by dietmarw