modelica/ModelicaStandardLibrary #2689: checkbox versus checkBox

tobolar, 2018-07-11:
`ModelicaReference.Annotations.choices` documents the usage of a _checkbox_ annotation. But there are used two variants of it in the text:
- checkbox and check**B**ox.

What is the correct one? I have checked Modelica Specification as well but there is the same ambiguity.
Dymola uses the upper case variant.

HansOlsson, 2018-07-11:
In Modelica 3.4 Specification that should be clear so that it is only checkBox spelling for the annotation; note that the text says "For a Boolean variable, a choices annotation may contain the definition checkBox = true, meaning to display a checkbox to input the values false or true in the graphical user interface."

The "checkbox" refers to the graphical element - https://en.wikipedia.org/wiki/Checkbox (alternative spellings are "check box", "tick box" and "tickbox").

https://trac.modelica.org/Modelica/ticket/1770

tobolar, 2018-07-11:
... so it should be updated in ModelicaReference as well.

commit a019ffb8 (2018-07-12): Fixes #2689  by correcting the spelling of the "checkBox" annotation (#2690)
closed 2018-07-12 by dietmarw