modelica/ModelicaStandardLibrary #2694: Add more spelling termes used in Electrical domain

christiankral, 2018-07-31:
In the Modelica documentation of `Modelica.UsersGuide.Conventions.Terms.Electrical` the spelling of terms are collected. This is a cumulative ticket to collect additional terms to be implemented in the future.

christiankral, 2018-07-31:
Add

    <tr>
      <td><a href="http://www.electropedia.org/iev/iev.nsf/display?openform&amp;ievref=351-47-17">set-point</a></td>
      <td>set point, setpoint</td>
    </tr>

christiankral, 2018-07-31:
<tr>
  <td><a href="http://www.electropedia.org/iev/iev.nsf/display?openform&amp;ievref=151-15-21">no-load</a></td>
  <td>noload, no load</td>
</tr>

christiankral, 2018-10-25:
```
<tr>
  <td><a href="http://www.electropedia.org/iev/iev.nsf/display?openform&amp;ievref=151-11-10">electromechanical</a></td>
  <td>electro mechanical, electro-mechanical</td>
</tr>
```

beutlich, 2019-03-25:
@christiankral Do you want to keep this issue open after #2854 got merged?

christiankral, 2019-03-25:
Yes, that's a good idea. Let's keep it open, as there will come more terms along...

christiankral, 2019-06-22:
How do you spell?

sub-package | subpackage | sub package

In the MSL I found sub-package and subpackage.

beutlich, 2019-06-22:
[subpackage](https://en.wiktionary.org/wiki/subpackage)