modelica/ModelicaStandardLibrary #2695: Storage of PNG files (figures)

tobolar, 2018-08-01:
There is a convention concerning figures' storage which states that
> The PNG files should be placed in a folder which exactly represents the package structure. 

(see also Modelica.UsersGuide.Conventions.Documentation.Format.Figures).

According to this, the figures collected in `Modelica/Resources/Images/UsersGuide` should be stored appropriately. This is already the case for figures used in Modelica.UsersGuide.Conventions, i.e. there is a directory `Modelica/Resources/Images/UsersGuide/Conventions` containing corresponding subdirectories.

tobolar, 2020-01-15:
@AHaumer @christiankral I'm going to proceed here with Electrical and Magnetic in the next days. In order to omit conflicts please let me know when I could start. Or you can of course fix this issue for those two packages on your own.

dietmarw, 2020-01-24:
As far as I can tell there are no more open PRs left that would conflict with changes to be done in Electrical and Magnetic

tobolar, 2020-01-24:
You are right. But working on this issue is really time consuming and I will not find time to fix this till MSL 4.0.0 fix.

dietmarw, 2020-01-24:
Yes. Hopefully @christiankral and @AHaumer can take care of those.

christiankral, 2020-01-26:
OK then. @AHaumer I guess we should always indicate in advance what packages we are working on to avoid redundant work.

beutlich, 2020-03-02:
> 
> 
> You are right. But working on this issue is really time consuming and I will not find time to fix this till MSL 4.0.0 fix.

Remvong milestone.