modelica/ModelicaStandardLibrary #2705: Initialization of ModelicaTest.MultiBody.Joints.SphericalSpherical2

henrikt-ma, 2018-08-22:
I'd like to apologize in advance in case I have missed something obvious in this model, but I don't understand why the model is expected to initialize with a the positive solution for the _x_ coordinate of the {{bodyBox}}.  All I can see is that the other two coordinates are supposed to initialize to 0 due to the {{freeMotionScalarInit}}, but I don't see any initialization at all for the _x_ coordinate.  As I understand it, this leads to an initial guess of 0 for the quadratic equation for satisfying the length constraint of the {{sphericalSpherical}}.  Since 0 is equally close to the negative and positive solutions, I don't see why one would expect to get the positive solution.

henrikt-ma, 2018-08-22:
Similar initialization is found in {{ModelicaTest.MultiBody.Joints.UniversalSpherical}}.

tobolar, 2018-08-27:
To my understanding, the following holds for `sphericalSpherical`:

    (1) `rRod_0 * rRod_0 - rodLength * rodLength = 0; // by constraintResidue`
    (2) `rRod_0 = frame_b.r_0 - frame_a.r_0;                 // by equation`
    (3) `rodLength = 1; // by parameter`

Additionally, there is a limitation due to the `freeMotionScalarInit`:

     (4) `sphericalSpherical.frame_b.r_0.start[2] = freeMotionScalarInit.r_rel_a_2.start;`
     (5) `sphericalSpherical.frame_b.r_0.start[3] = freeMotionScalarInit.r_rel_a_3.start;`
and

     (6) `sphericalSpherical.frame_a.r_0 = zeros(3);`

So the only free variable for the constraint equation (1) is `sphericalSpherical.frame_b[1]` so
that the constraint actually yields
     `sphericalSpherical.frame_b[1]*sphericalSpherical.frame_b[1] - 1 = 0;`
and finally
     `sphericalSpherical.frame_b[1] = sqrt(1);`

But actually, I have no idea about the preference for the square roots solution. Maybe principal square root, see https://en.wikipedia.org/wiki/Square_root, can make it?

henrikt-ma, 2018-08-28:
Thanks for a good summary of the equations involved.  It shows that we are on the same page.

Once you've reached (with `x` = `sphericalSpherical.frame_b[1]`),
{{{
x * x - 1 = 0;
}}}
you're left with a non-linear equation, and it is in their nature that there can be multiple distinct solutions.  Sometimes, one is only interested in the positive solution, sometimes only in the negative solution, and some models might even be jumping between the two due to re-initialized variables at events.  I really don't think that we should start listing particular non-linear equations and which of their solutions to pick.  The desired solution must be specified by giving an initial guess that is close enough to make a local non-linear solver go in the expected direction.

tobolar, 2018-08-31:
It means in this particular case that e.g. `freeMotionScalarInit.r_rel_a_1(start = -1)` should lead to the negative start position of `bodyBox`, shouldn't it? But there is a conflict with guess values set by `bodyBox` itself, which obviously has a priority due to the state selection (at least in Dymola). Trying `bodyBox.r_0(start={-1,0,0})`, the negative start position of `bodyBox` is indeed selected as desired.
(In this case the model only works when disabling models's asserts.)

henrikt-ma, 2018-09-03:
It's the same with SystemModeler; initialization needs to be on `bodyBox.r_0`.  Thus, it seems that both Dymola and SystemModeler would get the desired result for this model if `bodyBox.r_0` had a `start` value of `{1, 0, 0}`.  (Here, 1 is coincidentally both just a positive number that makes the non-linear local solver find the desired solution, and the actual value at solution.)

I suggest that this is how the example should be modified.

henrikt-ma, 2018-09-27:
Yes, just as guess values.

henrikt-ma, 2018-10-09:
Please note that this issue shouldn't be closed without also fixing `ModelicaTest.MultiBody.Joints.UniversalSpherical`.

tobolar, 2018-10-09:
I think there is no problem with it as you've opened a separate issue for UniversalSpherical: #2720
So there is actually no conflict and the issues can be solved separately.

henrikt-ma, 2018-10-10:
My bad.  I'm sorry.

henrikt-ma, 2019-05-08:
I'm double sorry.  My comment regarding `ModelicaTest.MultiBody.Joints.UniversalSpherical` was correct.  Issue #2720 doesn't address the initialization of `bodyBox.r0`.  Setting the same guess values as in `SphericalSpherical2` does the job.  Reopening.

beutlich, 2019-05-08:
@henrikt-ma Could you please create a PR. Thanks.

PR #2730 (2018-12-11): Set guess values of ModelicaTest.MultiBody.Joints.SphericalSpherical2
Close  #2705
commit 834227ef (2018-12-11): Set guess values of ModelicaTest.MultiBody.Joints.SphericalSpherical2 (#2730)

Close  #2705
closed 2019-05-28 by beutlich