modelica/ModelicaStandardLibrary #2716: Where to save PNG files

tobolar, 2018-09-25:
The convention on figures' format in the html documentation (`Modelica.UsersGuide.Conventions.Documentation.Format.Figures`) says:
> The PNG files should be placed in a folder which exactly represents the package structure.

Does this mean for a specific package that
1. figures used in components have to be saved in particular directories having the same names as components itself, or
2. figures of all package's components are saved in one directory which is called according to the package?

A simple example to demonstrate the difference:
Shall the figure _Butterworth.png_, used in `Modelica.Blocks.Continuous.LowpassButterworth`, be saved in directory
1. `~/Modelica/Resources/Images/Blocks/Continuous/LowpassButterworth/` or
2. just `~/Modelica/Resources/Images/Blocks/Continuous/`?

Refs #2695.

tobolar, 2018-10-30:
No feedback so far. I would personally vote for 1. and am going to force this in #2695 when there will be no opposite reaction.

HansOlsson, 2018-10-30:
To me it seems as if most of the leaf-directories will just contain one or two images.

If that is the case I would prefer ~/Modelica/Resources/Images/Blocks/Continuous/ButterWorth.png instead of ~/Modelica/Resources/Images/Blocks/Continuous/LowpassButterworth/ButterWorth.png
(basically I don't like repetitions).

Or possibly ~/Modelica/Resources/Images/Blocks/Continuous/LowPassButterWorth.png

beutlich, 2018-10-30:
I am fine with any of the three proposals. Just go ahead.

christiankral, 2018-10-30:
I am also in favor `~/Modelica/Resources/Images/Blocks/Continuous/ButterWorth.png` as we then have no more than ten files in the directory `~/Modelica/Resources/Images/Blocks/Continuous/`. I personally also think that it makes no sense have the same name in a directory and file.

dietmarw, 2018-10-30:
For the organisation of the image you should also take a possible future modularisation into account. So each possible stand-alone sub-library should have its own Image folder with no further detailed structure required.

tobolar, 2018-10-31:
I think the motivation for that rule in `Modelica.UsersGuide.Conventions.Documentation.Format.Figures` is to omit hundreds of images within one `~/Resources/Images/` directory. The other exterme would be to have hundreds of directories with just one image.
Regarding comments above there is clear preference for option 2.

closed 2018-10-31 by tobolar