modelica/ModelicaStandardLibrary #2719: Bug in the modified Eucken method of Modelica.Media.IdealGases.Common.Functions.thermalConductivityEstimate

finback-at, 2018-09-27:
I don’t have the original paper, but 3 linked documents below show that
the modified Eucken correlation is

        (Lambda * M)/(eta * Cv) = 1.32 + 177 * (R/Cv).

R is the universal gas constant and Cv is heat capacity at mol.  Units of them are  “J/(mol.K)” and this equation is a dimensionless factor.  

Last term is rewritten to,

        1.77*(R/Cv) = 1.77/(Cv/R) = 1.77/((Cp-R)/R) = 1.77/((Cp/R - 1.0)

Cp in Modelica.Media.IdealGases.Common.Functions.thermalConductivityEstimate is specific heat capacity “J/(kg.K)”  and should be converted. 

       Cp [J/(kg.K)] ->  Cp * data.MM [J/(mol.K)]

This last term should be

       1.77 / ((Cp * data.MM / Modelica.Constants.R) - 1.0).

Modified Eucken method equation in Modelica.Media.IdealGases.Common.Functions.thermalConductivityEstimate is

      eta*(Cp - data.R)*(1.32 + 1.77/((Cp/Modelica.Constants.R) - 1.0))

I think it should be corrected to,

      eta*(Cp - data.R)*(1.32 + 1.77/(((Cp*data.MM/Modelica.Constants.R) - 1.0)).

I checked this corrected equation for the cases of N2 and CH4 and those show better 
agreement with data of NIST Chemistry WebBook. 

Links,.
www.geocities.jp/penguinitis2002/study/note/gas_property.pdf eq. (16)
https://ws680.nist.gov/publication/get_pdf.cfm?pub_id=914943  eq. (11)
https://pdfs.semanticscholar.org/550c/f570367b409c3e947bac864dc4f88f2c86ba.pdf eq.(2)

Best regards
Amane Tanaka
amane@amane.to

beutlich, 2018-09-27:
Thank you for the issue report with concrete analysis. Can you please check, if this is already solved in MSL [v3.2.3-rc.1](https://github.com/modelica/ModelicaStandardLibrary/releases/tag/v3.2.3-rc.1), particularily by ef8359f75cf9ad6fdd071e3559023ebeffaacdf3 for #2377?

thorade, 2018-09-27:
In the function Modelica.Media.IdealGases.Common.Functions.thermalConductivityEstimate variables have following units:
Cp has unit J/kg.K
data.R has unit J/kg.K
Modelica.Constants.R has unit J/mol.K
lambda has W/m.K
eta has unit Pa/s

So, I think your analysis is correct, and the same equation can correctly be written in mass-based or mol-based units. Here, both are mixed incorrectly, and that happened because the same name R was used for mass-based and mol-based gas constant.
#2438 suggests to use R for mol-based universal gas constant and R_specific=R/MM for mass-based substance specific gas constant.

HansOlsson, 2018-10-01:
> Note @HansOlsson : Should Dymola 2019 unit checking catch this?

With default settings it will not be caught.  

The problem is that it is inside a function and you have the two constants 1.32 and 1.77 messing up the unit-checking (they are unit-less, but in general we are relaxed for such constants).

The tickets #2376 #2377 #2379 were created by being less strict, but not in functions.

If we also are less strict for functions it would be detected but so will all functions in Modelica.SIunits.Conversions. There is a proposal from Francesco about improving unit-checking, I will have to investigate if that would detect this issue and no others.

closed 2018-10-15 by beutlich

thorade, 2021-03-04:
@HansOlsson and @casella Could you point me to the proposal for better unit checking?

HansOlsson, 2021-03-04:
The original proposal with discussion was in https://github.com/modelica/ModelicaSpecification/issues/2127
However, the conclusion was to rework it - and little progress has been made.

As for the extended unit-checking possibilities in Dymola I would need time to dig up details on exactly what they are and what they do, but it might be that we should use that as a start - as it has found a number of tricky unit issues without finding too many false positives.

PR #2734 (2018-10-15): Issue2719 modified Eucken R in J/kg.K
This closes #2719.
```
data.R = Modelica.Constant.R/data.MM
```
@finback-at: This should be identical to the solution you suggested, could you verify it gives the same results?