modelica/ModelicaStandardLibrary #2720: Initialization of ModelicaTest.MultiBody.Joints.UniversalSpherical

henrikt-ma, 2018-09-27:
For the reference system in `ModelicaTest.MultiBody.Joints.UniversalSpherical`, there is a problem with the initialization of `universal.phi_a`.  In order for the `universal` component to do the pendulum motion in the correct plane, `universal.phi_a` must be 90°+-n*180°, but it is initialized to 0°.

I don't know why, but with SystemModeler, I find that -90° is a stable solution, while -90° is unstable but still seems to give the correct reference behavior.

The initialization at 0° thus leads to an infinitely short transition to -90° as the pendulum arm leaves the horizontal position.  In the current reference results, however, most of this transition happens before the second data point, at time 0.0004, but it takes about 0.0024 to reach a five digit approximation of π/2.  In the meantime, `universal.w_a` has a large negative peak that it doesn't even make sense to have reference values for.

The following alternatives could be possible solutions to this problem:
- Exclude at least `universal.w_a` from the comparison signals (`universal.phi_a` is less problematic since it has more or less the correct value at each data point).
- Initialize `universal.phi_a` to -90°.

tobolar, 2018-09-27:
Please correct which angle is stable:
>  -90° is a stable solution, while -90° is unstable

henrikt-ma, 2018-10-10:
I've corrected the original comment to say that 90° is unstable.

closed 2019-03-27 by beutlich

HansOlsson, 2019-03-27:
I'm not convinced that the problem was start-value of universal.phi_a, and think that it might be a modeling issue - in particular that the axis of the universal joint shouldn't be in the same direction as the following rod.

Consider the silly change of setting world.n={-0.1,-1,-0.1}; after that change universal.phi_a will then flip half-way through the simulation.

To me it more seems that universal.n_a should have been {0,0,1} instead of {0,0,1} (and then changing universal.phi_a back to zero), since the two universalSpherical have n1_a={0,0,1}. 

Shall I open another ticket for this? 
Since it is then a matter of incorrectly built model, not just initialization.

tobolar, 2019-03-27:
@HansOlsson I would agree, but I actually don't know the initial aim of the test model. Possible that `universal.n_a` is set in that way on purpose?

henrikt-ma, 2019-03-27:
That's a good analysis.  If the model is intentionally setting strange axes, one might wonder if it really makes sense to have a regression test for it.  This results in three alternatives to choose from:
- Model is intentionally setting strange axes, and is not well suited for regression testing; just delete the model.
- The strange axes is a modeling error; just fix the model.
- Model has bad intentions, but instead of deleting it, one could change the purpose of the model so that it becomes meaningful to keep; document the change of intent somewhere, and fix any modeling errors in view of the new intent.

I think it would make sense to open a new issue just for resolving which way to go here.

tobolar, 2019-03-27:
@HansOlsson 
>Shall I open another ticket for this?

@henrikt-ma 
> I think it would make sense to open a new issue just for resolving which way to go here.

I agree.

tobolar, 2019-05-13:
Similarly to SphericalSpherical2 (see https://github.com/modelica/ModelicaStandardLibrary/issues/2705#issuecomment-418034621), the initial guess value shall be also set for `bodyBox.r_0.start`. This is resolved with #2922.

PR #2731 (2019-03-27): Better initial angle phi_a of ModelicaTest.MultiBody.Joints.UniversalSpherical
Close #2720
commit efdbd782 (2019-03-27): Better initial angle phi_a of ModelicaTest...UniversalSpherical
Close #2720