modelica/ModelicaStandardLibrary #2721: ModelicaReference.Annotations.Dialog contains unresolvable URIs

svorkoetter, 2018-10-02:
The documentation in `ModelicaReference.Annotations.Dialog` contains two instances of `modelica://Multibody...`. I believe these should be `modelica://Modelica/Multibody/...` to resolve correctly.

sjoelund, 2018-10-03:
https://github.com/modelica/ModelicaStandardLibrary/blob/467a0b68f7f3fb455cda84583d6f6b20db9f45a8/ModelicaReference/package.mo#L760
is not a link though. It could refer to anything, and I suppose in this instance it referred to the old MultiBody library. I suppose this link should be `modelica://Modelica.Mechanics.MultiBody.UsersGuide.Tutorial` to resolve correctly (didn't check MSL to see the new location).

dietmarw, 2018-10-03:
Yes that location is correct. But it does not have to be an existing Package in this **example* snippet. I will supply a PR which changes this to a URI that does not indicate an existing package.

commit c0d3f3cf (2018-12-11): Fixes #2721 by using a more abstract package as example (#2723)

The previous example URI suggested something that should actually exist. But in this case this is just meant to serve as an example for documentation. This makes this clear.
closed 2018-12-11 by dietmarw