modelica/ModelicaStandardLibrary #2728: Remove comparison signal from regression test

beutlich, 2018-10-04:
The nextTimeEvent variable is a protected, discrete state variable, that should not be used as comparison signal of the ModelicaTest.Tables.CombiTimeTable tests, since it depends on Modelica.Constants.inf = ModelicaServices.Machine.inf, i.e., is tool-dependent (see #2435).

GallLeo, 2018-10-05:
Removed nextTimeEvent for all test cases in `ModelicaTest.Tables.*`

Ticket can be closed after upload of new reference results.

beutlich, 2018-10-15:
Please also remove ModelicaTest.MultiBody.PlanarLoopWithMove.PlanarLoopWithMove.combiTimeTable.nextTimeEvent.

GallLeo, 2019-01-14:
1. Removed ModelicaTest.MultiBody.PlanarLoopWithMove.PlanarLoopWithMove.combiTimeTable.nextTimeEvent
from my test setup, as requested.
2. I searched through all comparison signals and could not find string "nextTimeEvent" anymore.

This will be publiclty available, as soon as we create the final reference results based on v3.2.3.

closed 2019-01-14 by GallLeo