modelica/ModelicaStandardLibrary #2741: Regression in ModelicaTest.Tables.CombiTimeTable.Test6 ?

beutlich, 2018-10-14:
The t_new.y[1] signal of ModelicaTest.Tables.CombiTimeTable.Test6 misses the time event at 0.61 (+ period), if MSL v3.2.3-rc.1 (/w or w/o table fix of #2725 and Dymola 2019) is compared against MSL v3.2.2 (and Dymola 2014?). See http://www.ltx.de/download/MA/Compare_MSL_v3.2.3-rc.1/2018-10-06_202102_ModelicaTest_v3.2.3-rc.1_2019/compare/2018-10-06T21-28-00-Test6_report.html and in detail:

[image]

To reproduce run `Compare --tolerance 2e-3 --delimiter , --verbosity 2 Test6_t_new_y_1.csv Test6.csv`, where Test6_t_new_y_1.csv is the current result under investigation and Test6.csv is the official reference result provided by the MA.
Get Compare from https://github.com/modelica-tools/csv-compare,  Test6_t_new_y_1.csv from 
[Test6_t_new_y_1.zip](https://github.com/modelica/ModelicaStandardLibrary/files/2476736/Test6_t_new_y_1.zip) and Test6.csv from https://svn.modelica.org/projects/RegressionTesting/ReferenceResults/MSL/trunk/ModelicaTest/Tables/CombiTimeTable/Test6/.

I am not sure if this is a tool regression or library regression, since I am neither able to reproduce in SimulationX 3.9.4 nor to debug in Dymola 2019.

@HansOlsson Can you please investigate.
@GallLeo FYI

beutlich, 2018-10-15:
This is a tool regression in Dymola 2019 that can be resolved if Advanced.EfficientMinorEvents=false is set.

@GallLeo We need to decide if we
1. set this option in newer regression runs,
1. rerun MSL v3.2.2 regression in Dymola 2019 to get update reference results, or
1. simply accept all the false positives

to overcome the tool regression.

HansOlsson, 2018-10-16:
Since I'm assigned to this issue I will clarify as much as possible: this will be corrected in the next release; setting Advanced.EfficientMinorEvents=false is a work-around, the time-event isn't fully missed but it isn't stored in the result file. I believe that either option 1 or 3 above would be good.

GallLeo, 2018-11-07:
Advanced.EfficientMinorEvents=false is now the default setting for further release tests in Dymola 2019.