modelica/ModelicaStandardLibrary #2743: No medium2 in Modelica.Media.Examples.MoistAir

beutlich, 2018-10-15:
Documentation of Modelica.Media.Examples.MoistAir mentions medium2, but there is no such second medium in the example model.

https://github.com/modelica/ModelicaStandardLibrary/blob/d053d332d61fd754e44bf680a8f80bde76d24315/Modelica/Media/package.mo#L2424-L2430

dietmarw, 2018-10-26:
I looked at this and it seems the documentation of `MoistAir` belongs actually into `MixtureGases`. Then also the reference to `medium2` makes sense. I could have simply moved the documentation over but then I'm not sure what the documentation for `MoistAir` should look like. @hubertus65 any suggestions?

PS: Duplicates of these models also exist in ModelicaTest and should probably be updated at the same time.

HansOlsson, 2018-10-26:
> PS: Duplicates of these models also exist in ModelicaTest and should probably be updated at the same time

Hmm.. That made me realize something:

Why do we have duplicates?
If we only want to test ModelicaTest, and not examples in MSL why duplicate it instead of just making a model in ModelicaTest that extends from the example?

dietmarw, 2018-10-26:
Yes that would make much more sense.

beutlich, 2018-10-26:
These duplicates should be sorted out by #1053, #2542 and #2745.

beutlich, 2019-10-02:
> I looked at this and it seems the documentation of `MoistAir` belongs actually into `MixtureGases`.

No, it was copied from IdealGasH2O.

PR #3130 (2019-10-09): Remove wrong documentation of Modelica.Media.Examples.MoistAir
Closes #2743.
closed 2019-10-09 by hubertus65