modelica/ModelicaStandardLibrary #2750: American vs. BE

tobolar, 2018-10-23:


beutlich, 2018-10-23:
Indeed, currently it is neither defined nor consistently typed. I have no strong opinion here which flavour ehm flavor to use.

HansOlsson, 2018-10-23:
As I recall the difference between American and British English spelling (and grammar) is not always as clear-cut; and there are also other variations - like Canadian English.

Thus we might view enforcing this as being over-pedantic - or we might view it as marketing intended to make e.g. Americans welcome; or send some other message.

beutlich, 2018-10-24:
Even MLS 3.4 or FMI 2.0 specifications are not consistent (only checked spelling of behavior vs. behaviour). 

> Thus we might view enforcing this as being over-pedantic

I agree and would like to close this as non-issue.

tobolar, 2018-10-24:
I would appreciate to have some decission because a good work is marked by both the content and the form - imagine simply a scientific paper. But since I have no preference I agree to close this issue.

closed 2018-10-24 by tobolar

HansOlsson, 2018-10-24:
> Even MLS 3.4 or FMI 2.0 specifications are not consistent (only checked spelling of behavior vs. behaviour).

If you trust google it's complicated, and inconsistent.
The trends for behavior and colour are downward, and behaviour and color are steady/gaining ground (in both British and American English)! 

https://books.google.com/ngrams/graph?content=behavior%3Aeng_gb_2012%2Cbehaviour%3Aeng_gb_2012%2Cbehavior%3Aeng_us_2012%2C%2Cbehaviour%3Aeng_us_2012&year_start=1800&year_end=2000&corpus=15&smoothing=3&share=&direct_url=t1%3B%2Cbehavior%3Aeng_gb_2012%3B%2Cc0%3B.t1%3B%2Cbehaviour%3Aeng_gb_2012%3B%2Cc0%3B.t1%3B%2Cbehavior%3Aeng_us_2012%3B%2Cc0%3B.t1%3B%2Cbehaviour%3Aeng_us_2012%3B%2Cc0

https://books.google.com/ngrams/graph?content=color%3Aeng_gb_2012%2Ccolour%3Aeng_gb_2012%2Ccolor%3Aeng_us_2012%2Ccolour%3Aeng_us_2012&year_start=1800&year_end=2000&corpus=15&smoothing=3&share=&direct_url=t1%3B%2Ccolor%3Aeng_gb_2012%3B%2Cc0%3B.t1%3B%2Ccolour%3Aeng_gb_2012%3B%2Cc0%3B.t1%3B%2Ccolor%3Aeng_us_2012%3B%2Cc0%3B.t1%3B%2Ccolour%3Aeng_us_2012%3B%2Cc0

tobolar, 2018-10-24:
Seems the number of US books dealing with behavior decreases. :-)
Ok, you convinced me.