modelica/ModelicaStandardLibrary #2752: Use quoted section for preformated text

tobolar, 2018-10-23:
I suggest to quote sections of preformated text in the documentation, in HTML by `<blockquote> ... </blockquote>`. This is already established in plenty of classes, but not consequently in all classes.
If agreed, this should especially be documented in `Modelica.UsersGuide.Conventions.Documentation.Format.Code`.

tobolar, 2018-10-30:
I found out there is right opposite definition in `Modelica.UsersGuide.Conventions.Documentation.Format.Code`:
> 3. Multi line or single line code shall not be indented.

But I still believe the text in the second figure below is better readable than the text in the first figure and vote for changing this rule.
<hr>

**Figure 1** - snapshot of the current "Code" document:

> [image]

**Figure 2** - the same document, code lines formated using blockquote:

> [image]

beutlich, 2018-10-30:
Are the blockquote tags missing in the 2nd example of the 2nd figure? Otherwise I wonder why it is indented w.r.t. the 1st figure.

tobolar, 2018-10-31:
@beutlich Fixed 2nd figure.

beutlich, 2019-10-15:
@tobolar Seems accepted.

1. Can you please update the conventions! Thanks.
1. Do we also want to fix it in current MSL for v4.0.0?

tobolar, 2019-10-21:
> 2. Do we also want to fix it in current MSL for v4.0.0?

You mean to go through all &lt;pre&gt;s?

beutlich, 2019-10-21:
> 
> 
> > 1. Do we also want to fix it in current MSL for v4.0.0?
> 
> You mean to go through all <pre>s?

Yes.

tobolar, 2019-10-23:
OK, I will look over it.

PR #3171 (2019-12-17): Use quoted section for preformated text
@tobolar This is my brute-force search-and-replace. Not sure if it is good enough.

Resolves #2752.
closed 2019-12-17 by beutlich