modelica/ModelicaStandardLibrary #2757: Very bad behaviour of two Blocks.Logical icons (LessThreshold and GreaterThreshold)

max-privato, 2018-10-26:
The icon of LessThreshold is a sign "<". When the component is used as is, it is meaningful.
If the component flipped horizontally, the same icon "<" is displayed, which indicates greater instead of less!

Also GreaterThreshold is affected by this issue. The issue can be solved using a drawing for the less or greater symbol instead of a character. 

Less, Greater, LessEqual and GreaterEqual are unaffected, since their icons are already drawings.

beutlich, 2018-10-28:
Left over from #745. Will prepare a pull request.

max-privato, 2018-11-02:
Thank you for implementing my suggestion.
After some usage, and getting other people involved, I came to the conclusion that the icon is still confusing. The issue is that when we use the symbol "<" we want to have two terms In Greater and Less somehow we have them, in GreaterThreshold, and LessThreshold, even with the modification i suggested we still don't have.

I made some tests with both Dymola and Openmodelica, and using the text "Th" as a placeholder for threshold seems good enough, and well coordinated with the existing Less and Greater icons.
The annotation code for LessThreshold that I tested (for me satisfactorily) is as follows (excluding Documentation):

 annotation (Icon(coordinateSystem(preserveAspectRatio=true,
  extent={{-100,-100},
         {100,100}}), graphics={
          Line(points={{-36,20},{-82,0},{-36,-20}},
          thickness=0.5), Text(
          extent={{-26,48},{46,-48}},
          lineColor={0,0,0},
          textString="Th",
          horizontalAlignment=TextAlignment.Left)}),
          [...]

@beutlich, do you agree with this further enhancement? If you do, obviously also GreaterThreshold should be changed accordingly.

beutlich, 2018-11-02:
> @beutlich, do you agree with this further enhancement?

Forwarding to library officers @AHaumer and @MartinOtter .

PR #2758 (2018-12-11): Fix icons to utilize graphics instead of text
Ensure consistent looks of all 8 affected icons

close #2757
commit dc0d9342 (2018-12-11): Fix icons to utilize graphics instead of text (#2758)

Ensure consistent looks of all 8 affected icons

close #2757
closed 2018-12-11 by dietmarw