modelica/ModelicaStandardLibrary #2761: Missing component descriptions in example ShowTransferFunction

beutlich, 2018-10-29:
Parameters a, b, d, wMax, wMin and variables dB, lg_w and phi miss a descriptive string.

See also #2166 and #1850.

AHaumer, 2018-11-05:
d: damping coefficient
b: numerator polynomial coefficients of transfer function
a: denominator polynomial coefficients of transfer function
wMin: lower bound for frequency sweep
wMax: upper bound for frequency sweep
lg_w: logarithm of frequency
dB: absolute of transfer function in decibel
phi: argument of transfer function

beutlich, 2018-11-05:
Thanks.

> dB: absolute of transfer function in decibel

Modified to magnitude in #2767. Please review.

beutlich, 2018-11-08:
> 
> 
> Please mark this also in #2636, if finished!

Done.

PR #2767 (2018-11-08): Add component descriptions to example Modelica.ComplexBlocks.Examples.ShowTransferFunction
close #2761
closed 2018-11-08 by beutlich