modelica/ModelicaStandardLibrary #2762: Output port for Multiphase sensors has wrong orientation

JoeRiel, 2018-10-30:
The output port for the Current, Power, and Voltage sensors in Modelica.Electrical.MultiPhase,Sensors  has orientation 90 rather than -90.  As such it appears to be an input rather than an output.

christiankral, 2018-10-30:
Example: For the current sensor the real output is:
```
  Modelica.Blocks.Interfaces.RealOutput i[m]
    "current in the branch from p to n as output signal" annotation (
      Placement(transformation(
        origin={0,-110},
        extent={{10,-10},{-10,10}},
        rotation=90)));
```

The `rotation` being `90` instead of `-90` is due to the `extent` being `{{10,-10},{-10,10}}` instead of `{{-10,-10},{10,10}}`. Both in OpenModelica and Dymola the output connector looks correct:

[image]

I do, however, not know if the Modelica specification of `extent` allows what we have currently implemented. Can someone of the language group make a comment here?

JoeRiel, 2018-10-30:
Thanks for the response. I didn't notice the flipped extent, which should correct the orientation.

beutlich, 2018-11-01:
Flipped extents are legal.

closed 2018-11-01 by beutlich