modelica/ModelicaStandardLibrary #2763: Read combiTimeTable using column header names

Mathadon, 2018-10-31:
The `CombiTimeTable` currently supports indexing using indices through the parameter `columns`. It would be useful if it were possible to index a header using header names, e.g. through parameter

```
parameter String[:] columnNames;
```
When a table structure/order changes, the `CombiTimeTable` then automatically adjusts as long as some header in the file is consistent with the data. I can propose an implementation if this proposal is approved.

beutlich, 2018-10-31:
What do you mean by header?

Mathadon, 2018-10-31:
I propose to add an optional line to the data file that contains a name for each column. E.g:

```
#1                                                                                                                                                                                                                                    
double tab1(389,30)
#HEADER
# col1       col2       another_column       ...       etc
1538773200.000000       285.320000      282.450000      0.830000        101347.000000   0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        173.000000      2.710000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000
1538774100.000000       285.212500      282.320000      0.827500        101340.000000   0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        173.250000      2.687500        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000
1538775000.000000       285.105000      282.190000      0.825000        101333.000000   0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        173.500000      2.665000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000        0.000000
```

beutlich, 2018-10-31:
You can already do now:
```mo
model CombiTimeTable2763
  extends Modelica.Blocks.Sources.CombiTimeTable(
    final tableOnFile=true,
	final columns = getColumns(fileName, tableName, columnNames));
  parameter String columnNames[:] = {"col1", "col2"} "Column names" annotation(Dialog(group="Table data interpretation"));
  function getColumns "Get columns from column names"
    input String fileName "File name";
    input String tableName "Table name";
	input String columnNames[:] "Column names";
	output Integer columns[size(columnNames, 1)] "Columns of table to be interpolated";
	external "C" TODO(fileName, tableName, columnNames, columns);
  end getColumns;
end CombiTimeTable2763;
```

Mathadon, 2018-10-31:
That is indeed a possible implementation. Just to make sure I understand this correctly: the function `TODO` is not implemented yet in the code above? And it would make sense to actually integrate that code snippet into the MSL?

beutlich, 2018-11-01:
Have a look at https://github.com/modelica/ModelicaStandardLibrary/blob/71590c7e658a6fbbe037364f52b82589b01a1003/Modelica/Resources/C-Sources/ModelicaIO.c#L653-L654

Also, column names are not available for MAT files. So, headers would only be valid for text based files and we would require a specialized class TxtTimeTable then.

Mathadon, 2018-11-01:
Good point, forgot about that. I better skip this for now then. Thanks for the feedback!

closed 2018-11-01 by Mathadon

sjoelund, 2018-11-01:
Why would you need the function to be an external function? If you can get the columns from the file, you can calculate the indexes by a lookup function.
```mo
final columns = getColumns(fileName, tableName, columnNames));
parameter String columnNames[:] = {"col1", "col2"} "Column names" annotation(Dialog(group="Table data interpretation"));
Integer indexes[:] = array(indexOf(name, columns) for name in columnNames);
```

beutlich, 2018-11-01:
> Why would you need the function to be an external function? If you can get the columns from the file, you can calculate the indexes by a lookup function.

Not possible to swap columns then: {"col2", "col1"}