modelica/ModelicaStandardLibrary #2764: Unit conversion bug in Media.IdealGases.Common.MixtureGasNasa.mixtureViscosityChung

finback-at, 2018-11-01:
The output value of mixtureViscosyChng for CombustionAir is in the order of 1e+9. Too large!
I checked the original document.
The unit of calculated value of etam is 

    (micro Poise) = 1e-7 (Pa.s)

Current output value is

`
   etaMixture := etam*1e7;
`

It should be corrected to,

`
etaMixture := etam*1e-7;
`

beutlich, 2018-11-01:
@thorade Are you familar with https://github.com/modelica/ModelicaStandardLibrary/blob/71590c7e658a6fbbe037364f52b82589b01a1003/Modelica/Media/IdealGases/Common/package.mo#L1061 ?

thorade, 2018-11-01:
Sorry, I have not used that correlation myself. The three references given are:
Chung 1988 https://doi.org/10.1021/ie00076a024
Chung 1984 https://doi.org/10.1021/i100013a002
general reference, book by Poling https://scholar.google.com/scholar?cluster=5190098174187723396

The unit used by default in MSL is Pa.s which is inconvenient for viscosity, a displayUnit µPa·s would be nice, also see #545.
https://en.wikipedia.org/wiki/Viscosity#Units

PR #2778 (2018-12-10): Fix conversion from microPoise to Pa.s in mixtureViscosityChung
@finback-at Thanks for reporting the bug. I could reproduce it using a tester https://gist.github.com/beutlich/6f382895be633f5aae27b5a621f4cfd2 and the [paper](https://doi.org/10.1021/ie00076a024 ) by Chung.

close #2764
closed 2018-12-10 by beutlich