modelica/ModelicaStandardLibrary #2765: ComplexInput and ComplexOutput icons have strange greyed boxes

max-privato, 2018-11-02:
ComplexBlocks.Interfaces.ComplexInput and ComplexBlocks.Interfaces.ComplexOutput icons Have a strange greyed background (a rounded box below the arrow).

I don't know whether this is done on purpose, but it seems to me rather confusing. 
I propose to remove those greyed boxes.

max-privato, 2018-11-02:
I see. Thank you for the explanation.
So I understand that we must live with those grey boxes, for an unpredictable (probably very long) time.
I posted the ticket just in case it was an easy fix, which resulted not to be the case.

max-privato, 2018-11-03:
> Probably, the easiest solution is to remove the icon graphics from the operator record (which is not displayed in the Dymola library panel anyway to pretend a builtin type Complex).

A fallback option is just adding an intermediate rectangle between the grey and the triangle as follows. Does not solve the issue in its roots, but hides it, I suppose satisfactorily.

connector ComplexOutput = output Complex "'output Complex' as connector"
  annotation (
  defaultComponentName="y",
  Icon(coordinateSystem(preserveAspectRatio=true, extent={{-100,-100},{100,
          100}}),
          graphics={Rectangle(
        extent={{-100,102},{100,-100}},
        fillColor={255,255,255},
        fillPattern=FillPattern.Solid,
        pattern=LinePattern.None),
                    Polygon(
        points={{-100,100},{100,0},{-100,-100},{-100,100}},
        lineColor={85,170,255},
        fillColor={255,255,255},
        fillPattern=FillPattern.Solid)}),
  Diagram(coordinateSystem(preserveAspectRatio=true, extent={{-100,-100},{
          100,100}}), graphics={Polygon(
        points={{-100,50},{0,0},{-100,-50},{-100,50}},
        lineColor={85,170,255},
        fillColor={255,255,255},
        fillPattern=FillPattern.Solid), Text(
        extent={{30,110},{30,60}},
        lineColor={0,0,127},
        textString="%name")}),
  Documentation(info="<html>
<p>
Connector with one output signal of type Complex.
</p>
</html>"));

beutlich, 2018-11-03:
> A fallback option is just adding an intermediate rectangle between the grey and the triangle as follows. 

Possible. But not good enough for a _standard_ library.

max-privato, 2018-11-03:
>Possible. But not good enough for a standard library.

Yes it is tricky. In addition, it works well with Dymola, but is unreliable in OM.

max-privato, 2018-11-03:
I clarify a little.
In case it turns out that because of a hole in Modelica specification no cleaner solutions exist, maybe a tricky -one is acceptable also in MSL? 
That's why I called it "a fallback option".

But such a choice should be painless for users. If does not work in OM it is unacceptable. 
I will try to understand why. In case there is an easy-to-solve issue OpenModelica code I can open a ticket on this.

HansOlsson, 2018-11-07:
> @HansOlsson Do you confirm the language hole?

To me the simplest way of specifying it would be:
`connector ComplexOutput=output Complex annotation(IconMap(primitivesVisible=false),Icon(...));`

As far as I can understand that should work according to 18.6: 
"The allowed annotations for a short class definition is the union of the allowed annotations in classes and on extends-clauses."
But we could likely clarify that further, and e.g. show it as an example.
(Additionally, it does not seem to be implemented in all tools.)

beutlich, 2018-11-07:
> To me the simplest way of specifying it would be:
> `connector ComplexOutput=output Complex annotation(IconMap(primitivesVisible=false),Icon(...));`

Ok.

The other option would be to allow base-prefix for long-class definitions., which opens the general question if every short-class defintion must always be possible to be defined as long-class equivalent.

beutlich, 2018-11-07:
> (Additionally, it does not seem to be implemented in all tools.)

Afaik, SimulationX, OMEdit and MapleSim support the IconMap. It's just that it is not yet used in the MSL and that there are some open clarification issues to be dealt with.

HansOlsson, 2018-11-07:
> The other option would be to allow base-prefix for long-class definitions., which opens the general question if every short-class defintion must always be possible to be defined as long-class equivalent.

Since you cannot combine the base-prefixed class with any other contents that seems like something that is easy to use incorrectly.

> Afaik, SimulationX, OMEdit and MapleSim support the IconMap. It's just that it is not yet used in the MSL and that there are some open clarification issues to be dealt with.

I specifically meant IconMap on short class.

beutlich, 2018-11-07:
Right, got it. Will open a specification issue with your clarification proposal and a new MSL PR making it a tool issue then.

PR #2769 (2019-07-15): Set inherited Complex record icon invisible
This is an alternative pull request to #2768.

close #2765
closed 2019-07-15 by beutlich